microsoft/AirSim #536: problems about recording in the latest version of airsim

cca2016, 2017-10-12:
when I click the button to record scenes, the program successfully runs but there is only one picture in my documents/airsim always.
I do not know what happened.
And I am not very clear about the usage of API. Can you show me how to use APIs?
Here are my settings.json:
{
  "SettingsVersion": 1.0,
  "DefaultVehicleConfig": "",
  "SimMode": "Car",
  "ClockType": "",
  "ClockSpeed": 1,
  "LocalHostIp": "127.0.0.1",
  "RecordUIVisible": true,
  "LogMessagesVisible": true,
  "ViewMode": "",
  "UsageScenario": "",
  "RpcEnabled": true,
  "EngineSound": true,
  "PhysicsEngineName": "",
  "EnableCollisionPassthrogh": false,
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05
  },
  "CaptureSettings": [
    {
      "ImageType": 1,
      "Width": 256,
      "Height": 144,
      "FOV_Degrees": 90,
      "AutoExposureSpeed": 100,
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    }
  ],
  "SubWindows": [
    {"WindowID": 0, "CameraID": 0, "ImageType": 3, "Visible": true},
    {"WindowID": 1, "CameraID": 0, "ImageType": 5, "Visible": true},
    {"WindowID": 2, "CameraID": 0, "ImageType": 0, "Visible": true}    
  ] ,
  "SimpleFlight": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Armed",
    "RC": {
      "RemoteControlID": 0,
      "AllowAPIWhenDisconnected": false,
      "AllowAPIAlways": true
    },
    "ApiServerPort": 41451
  },
  "PX4": {
    "FirmwareName": "PX4",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "VehicleCompID": 1,
    "VehicleSysID": 135,
    "ApiServerPort": 41451
  }
}

sytelus, 2017-10-13:
I can't reproduce your issue. In current version of AirSim, images are generated as expected. Please see docs for API usage: https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md

closed 2017-10-13 by sytelus

oppne, 2017-10-24:
I have a similar issue, no matter how much i change the settings for the width and height, it doesnt change it for the actual pictures thats the recording.