microsoft/AirSim #537: How to record three typies of image ?

cca2016, 2017-10-13:
Here is my JSON:
{
  "SettingsVersion": 1.0,
  "DefaultVehicleConfig": "",
  "SimMode": "Car",
  "LocalHostIp": "127.0.0.1",
  "ViewMode": "",
  "UsageScenario": "",
  "RpcEnabled": true,
  
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05
  },
  "CaptureSettings": [
    {
      "ImageType": 0,
      "Width": 256,
      "Height": 144,
      "FOV_Degrees": 90,
      "AutoExposureSpeed": 100,
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    },
	{
      "ImageType": 3,
      "Width": 256,
      "Height": 144,
      "FOV_Degrees": 90,
      "AutoExposureSpeed": 100,
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    },
	{
      "ImageType": 5,
      "Width": 256,
      "Height": 144,
      "FOV_Degrees": 90,
      "AutoExposureSpeed": 100,
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    }
  ],
  "SubWindows": [
    {"WindowID": 0, "CameraID": 0, "ImageType": 3, "Visible": true},
    {"WindowID": 1, "CameraID": 0, "ImageType": 5, "Visible": true},
    {"WindowID": 2, "CameraID": 0, "ImageType": 0, "Visible": true}    
  ] ,
  "SimpleFlight": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Armed",
    "RC": {
      "RemoteControlID": 0,
      "AllowAPIWhenDisconnected": false,
      "AllowAPIAlways": true
    },
    "ApiServerPort": 41451
  },
  "PX4": {
    "FirmwareName": "PX4",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "VehicleCompID": 1,
    "VehicleSysID": 135,
    "ApiServerPort": 41451
  }
}

But I found that this setting could only record scene picture, can you help me?
Thanks in advance

sytelus, 2017-10-13:
How are you recording? The record button (or R key) currently only records scene image type. If you want other image types then you should use Python or C++ APIs. You don't need to create CaptureSettings for that (default values will be used).

cca2016, 2017-10-15:
Yeah, I only used the red button. In fact  I am not quite familiar with the usage of API. Can you give me an example? Where to find the way to use API?

sytelus, 2017-10-19:
https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md

closed 2017-10-19 by sytelus

sytelus, 2017-12-12:
The feature to record images from multiple cameras simultaneously is now available. You can, for example, record scene and depth images simultaneously. 

Please see the doc: https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#recording