microsoft/AirSim #538: simGetPose() & simSetPose() not working properly

cangokalp, 2017-10-13:
#532 - might be relevant

```
(Pdb) client.simGetPose()
{b'position': {b'x_val': 0.0, b'y_val': 0.0, b'z_val': 0.0}, b'orientation': {b'x_val': 0.0, b'z_val': 0.0, b'y_val': 0.0, b'w_val': 1.0}}
(Pdb) client.getPosition().z_val
-10.334099769592285
(Pdb) home
{b'position': {b'x_val': 0.0, b'y_val': 0.0, b'z_val': 0.0}, b'orientation': {b'x_val': 0.0, b'z_val': 0.0, b'y_val': 0.0, b'w_val': 1.0}}
(Pdb) client.simSetPose(home,True)
(Pdb) client.getPosition().z_val
-10.334085464477539
(Pdb) client.simGetPose()
{b'position': {b'x_val': 0.0, b'y_val': 0.0, b'z_val': 0.0}, b'orientation': {b'x_val': 0.0, b'z_val': 0.0, b'y_val': 0.0, b'w_val': 1.0}}

```

simGetPose() won't return the correct Pose, simSetPose() make the drone respawn from the starting point (given starting point Pose) then it flies to where it was before the simSetPose() command

sytelus, 2017-10-13:
Are you using these APIs in multirotor mode? They are not designed for use in multirotor mode. They should only be used in[ "Computer Vision" mode](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#computer-vision-mode).

cangokalp, 2017-10-18:
ah, gotcha

sytelus, 2017-10-19:
BTW, I did some refactoring so this API is now usable in other modes as well. Source is updated and new binaries will be updated soon as well.

closed 2017-10-19 by sytelus