microsoft/AirSim #539: Depth Map Issue when camera angle changed

Mayankm96, 2017-10-14:
For the purpose of my project, I have to set the camera angle such that it faces downwards. I did so by changing the camera orientation from the blueprint. However, I am noticing a black solid line in the depth map that is being generated now by the simulator which wasn't prominent earlier. It might be this is a bug due to some recent changes? Would love to know your opinion on this.

sytelus, 2017-10-19:
We just noticed this issue as well! I'm still not sure of cause but I'll send update in this thread.

Adding pan-tilt is fairly easy. Basically you want to add setCameraOrientation API. The API call will eventually call the SetActorRelativeRotation() on PIPCamera. However navigating through RPC infrastructure would be bit daunting if you haven't done before. In that case wait out few weeks while I implement this.

sytelus, 2017-10-19:
It appears that this issue is likely due to bug in Unreal. For now I have done workaround by switching the visualization from planner depth to perspective depth. This should be ok for visualization purposes.

PS: binaries are not updated yet, so please use latest source.

closed 2017-10-19 by sytelus

Mayankm96, 2017-10-21:
Thank you for your response! 

However, does it mean that it's only an issue with the visualization for the planar depth image? I would also like to know if it means that the pfm matrix for the depth contains the right values for the depth and whether it is a concern or not?

wells-sde, 2017-10-26:
@Mayankm96 Depth image is ok,you can just save it to check the result.A png format depth image is better for checking.

guillem74, 2018-03-15:
@Mayankm96 You would be so kind if you tell me how did you changed the camera angle so it faces downwards. For a Machine Learning project I need to face the camera downwards for the drone in the Neighborhood scenario.