microsoft/AirSim #542: Simulating wind using radial force feature of unreal engine

alvikag, 2017-10-16:
Hi,
I was wondering if its possible to apply wind to the simulated vehicle. As a first step, I added a radial force in the unreal environment. I am able to see the quadrotor drifting as a result of this, but this drift/change in position is not being reflected in the trajectory recorded using Airsim
I was wondering if this can be achieved by some modification,  because vehicle collisions that happen on the unreal side are recognized by the Airsim physics engine.

sytelus, 2017-10-19:
AirSim uses different physics engine so I can see why its internal state is not effected. Did you added global radial force? This is an open feature we want to implement however we would probably not use radial force in Unreal. Instead, we will define wind vector at each point in space-time and then compute force on vehicle using box model by doing dot product of wind vector and surface of the box and using pressure formula. This is in plan but it will take some time for us to get to implementation. If you know how to do this, we will be happy to have contribution from you :).

cangokalp, 2017-10-25:
This would be really nice to have in the simulation!

zaddan, 2017-11-08:
plus one for the wind! It'd be absolutely priceless to have wind!

sytelus, 2018-02-13:
Duplicate: https://github.com/Microsoft/AirSim/issues/284

Let's continue discussion in above issue.

closed 2018-02-13 by sytelus