microsoft/AirSim #543: Question Regarding YawMode

senakicir, 2017-10-16:
Hello! 

I am running one of the sample scripts, box.py. I found that for yaw=90 and yaw=180, the drone does not stay fixed at the angles 90 and 180, but continuously spins... However for yaw=270 and yaw=0, it stays fixed to a certain angle. Can you explain why this is so? Is this a bug?

I am asking because I am working on my own program which needs the drone to point toward a certain angle as it is flying. I am trying to understand how exactly YawMode(False, angle) works. 

Thanks!

Kjell-K, 2017-10-16:
You have to experiment with [`drivetrain` and` yaw_mode`](https://github.com/Microsoft/AirSim/blob/2eb0e4934dac4ff6eeb2bbb562581426f0ec8bd8/docs/apis.md).

> **drivetrain**
> 
> Many APIs take drivetrain parameter which specifies how vehicle is oriented while it moves. For example, a quadrotor may move in West direction while its front is pointing towards North. In fact, quad rotors can move in all 3 axis while its front is pointing to any direction. In this mode of movement, vehicle uses its maximum degree of freedom. In forward-only mode, vehicle move only in the direction its pointing to. So if quadrotor is pointing to North and if we want it to move along West then we must reorient it. This mode is useful if we have only one sensor in the front.
> 
> **yaw_mode**
> 
> Many APIs take yaw_mode parameter which specifies where vehicle should be pointing to. For example, you may want to move quadrotor while its front is pointing to North-East which means you will set YawMode::is_rate = false and YawMode::yaw_or_rate = 90 (angle unit is degrees for Yaw). In some weired world, you might want your vehicle to continuously rotate at, say 20-degrees/sec, while it moves. YawMode allows you to specify that by setting YawMode::is_rate = true and YawMode::yaw_or_rate = 20. However in more sane scenarios, you just don't want yaw to change which you can do by setting yaw rate of 0. The shothand for this is YawMode::Zero().

senakicir, 2017-10-16:
Hello! Thank you for your reply. I already played with those, I'm setting drivetrain to DrivetrainType.MaxDegreeOfFreedom, so the movement is not restricted by the direction the drone is facing. 
What I am puzzled by is yaw_mode itself, I feel that it does not behave as it should. Have you tried running box.py? Is it behaving erratically for you too, for degrees 90 and 180?

sytelus, 2017-10-19:
YawMode is struct with two fields, aw_or_rate and is_rate. If is_rate = true then yaw_or_rate is interpreted as angular velocity in degrees/sec. If is_rate = false then yaw_or_rate is interpreted as angle in degrees.

How does this get used?

There are two modes you can fly vehicle: Drivetrain = ForwardOnly and MaxDegreeOfFreedom. When you specify ForwardOnly, you are saying that vehicle's front should always point in the direction of travel. So if you want drone to take left turn then it would first rotate so front points to left. This mode is useful when you have only front camera and you are operating vehicle using FPV view. The MaxDegreeOfFreedom means you don't care where the front points to. So when you take left turn, you just start going left like crab.

Now you can probably see that YawMode::is_rate = true is pointless in ForwardOnly mode because you are contradicting by saying that keep front pointing ahead but also rotate.  However if you have YawMode::is_rate = false in ForwardOnly then you can do some funky stuff. For example, you can have drone do circles and have YawMode::yaw_or_rate = 90 so camera is always pointed to center ("super cool selfie mode"). In MaxDegreeofFreedom, you can get some funky stuff by setting YawMode::is_rate = true and say YawMode::yaw_or_rate = 20. This will cause drone to go in its path while rotating which may allow to do 360 scanning.

Having understood above, please let me know if you still see unexpected behavior.

senakicir, 2017-10-20:
Hello! Thank you for the very detailed explanation, it was very helpful! I managed to make my program work exactly how I wanted it to.

I was confused by YawMode::is_rate = false and setting yaw_or_rate to a degree (let's say x). I had thought the drone would rotate so the yaw angle would be equal to x. If I understood correctly, it actually _increments_ the yaw angle by x. (Actually, it incremented it by x*pi degrees so I am passing it x/pi.)

Thank you for all your help!

senakicir, 2018-02-19:
Hello! I saw that you fixed the continuous rotation issue in a commit made back in December 16. (specifically, commit f7ab5ae0844c0ffceff52e10235869a04aac797a). Unfortunately this broke my workaround, so I'm trying to readjust my algorithm to suit your new fixed version. 

I'm trying to use the MaxDegreesOfFreedom mode, with is_rate = False. I know exactly what angle I want my drone to be facing. The problem that I have is, when I'm at degree 175 and I want to rotate to -175, instead of turning just 10 degrees, the drone does a 350 degree turn. Similarly, when it goes from 8 to 358 degrees, instead of just turning 10 degrees, it turns 350 degrees.

So in short, I do not understand how you can control the direction you want to rotate with YawMode. Is there a way to control this? 

Thank you.

mescarra, 2018-03-22:
I'm having a similar behaviour. I'm using the HelloDrone example (square path), and only changing the yaw_mode.is_rate to false, as @sytelus suggests. This works fine the first two corners, and the drone rotates 90º to move forward to the next corner, but then the third corner makes the drone spin without stop. Cannot figure out why this happens ... Any ideas?

sytelus, 2018-04-12:
We had a bug that caused yawing unnaturally for the drone. It's fixed now.

madratman, 2019-02-19:
closing as this was resolved. however, yaw_mode needs some more elucidation in the docs, specifically shital's comment https://github.com/Microsoft/AirSim/issues/543#issuecomment-337830728

closed 2019-02-19 by madratman