microsoft/AirSim #545: Running hello_drone.py, RemoteControl switch posision disallows API control

HuyenPhuongVu, 2017-10-17:
Hi, I am new to AirSim. 
I have been trying to run the example hello_drone.py but it gives the error "RemoteControl switch posision disallows API control, requestApiControl failed". 
I change the settings.json file to "computer vision mode". As I understand in the computer vision mode, I can use the arrow key to move around and take the images as in the example code.
Can you please help me figure out what is wrong, is there any additional setup that I need to do to make the example run correctly?

Thanks,
Bests,
Huyen

ShreyasSkandan, 2017-10-18:
I've been having the same problem on Windows 10 64bit.

sytelus, 2017-10-19:
Do you have any joystick, XBox controller etc connected? I think what is happening is that AirSim is detecting some joystick and some switch on that joystick is set to on position. AirSim uses safety protocol just like DJI Matrice, i.e., operator must allow API control by setting RC to "F-Mode". This prevents situation where commands coming through APIs might be unplanned and cause safety issue. When nothing is connected, AirSim allows API control by default.

HuyenPhuongVu, 2017-10-19:
Thanks for reply, I was thinking the same, maybe something was detected as joystick (I don't have any XBox controller connected, I have only a dock connected , is it considered as a kind of joystick or controller?). But as I run the hello_car.py, everything is fine.
Also, if I run DowntownCar.exe (with the multirotor) only (without running hello_drone.py) it says on the screen "RC controller on USB not detected, API server started at...", so I think no RC controller detected.

Please let me know if you have any clue.

Thanks,
Bests,
Huyen

JM23849, 2017-10-19:
I am experiencing the same issue, which only occurs when the UsageScenario is set to ComputerVision mode (I don't have any joysticks, controllers etc. connected). The API control for the C++ HelloDrone.exe works perfectly fine in SimpleFlight mode, but once I change the scenario to ComputerVision I can't get API to work regardless of the settings I change.

The full message in the top left of Unreal once I try to control via API in computer vision is below:
(standard blue messages:)
Default config: SimpleFlight
Press F1 to see help
RC Controller on USB not detected: 4294967295
API server started at (default)

("error" messages in green:)
RemoteControl switch position disallows API control
requestApiControl failed
Vehicle cannot be armed via API because API has not been given control
requestApiControl() must be called before using API control)

client.confirmConnection(); actually works, as I get a message "Connected!" in the console application for HelloDrone.exe . After HelloDrone is connected, the first three green messages above appear. Once HelloDrone tries to send its first actual command to the drone, the fourth green message appears ("requestApiControl() must be called...").

Below are the settings I am using in the .json file when I get API to work for basic SimpleFlight:

`{
  "SimpleFlight": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Inactive",
    "RC": {
      "RemoteControlID": 0,
      "AllowAPIWhenDisconnected": true,
      "AllowAPIAlways": true
    },
    "ApiServerPort": 41451
  },
  "SettingsVersion": 1.0
}`

And below are the settings I used for ComputerVision mode, though no variation of settings for ComputerVision mode can get this to work (including the "RC" portion above does not help):

`{
  "SimpleFlight": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Inactive",
    "ApiServerPort": 41451
  },
  "SettingsVersion": 1.0,
  "UsageScenario": "ComputerVision"
}`

I only just tried to control the drone via API in ComputerVision mode after the latest update, but I was under the impression that it was already supported (as suggested by issue #357 , as it looks like gawela was able to get it working). Being able to control the drone via rpc API in ComputerVision mode (physics turned off) would be very valuable, so I hope it is supported.

Let me know if you need any more information.

HuyenPhuongVu, 2017-10-19:
Thanks for reply, I am confused a bit, do you get it work by setting to 

`{
  "SimpleFlight": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Inactive",
    "RC": {
      "RemoteControlID": 0,
      "AllowAPIWhenDisconnected": true,
      "AllowAPIAlways": true
    },
    "ApiServerPort": 41451
  },
  "SettingsVersion": 1.0
}`
or 
`{
  "SimpleFlight": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Inactive",
    "ApiServerPort": 41451
  },
  "SettingsVersion": 1.0,
  "UsageScenario": "ComputerVision"
}`
Because I tried both but it still gives the same error when I run the hello_drone.py.

Thanks,
Bests,
Huyen

JM23849, 2017-10-19:
I cannot get it to work, sorry if there is any wording in the post above that makes it seem like I could. So long as the UsageScenario is ComputerVision, no difference in the settings.json file will allow me to get the API to work (I just get the same 4 green messages about the API failing as shown above).

HuyenPhuongVu, 2017-10-20:
Ok, I think I have exactly the same problem as you when I used the ComputerVision mode.
But I think in the hello_drone.py, we have to use the ComputerVision mode to navigate the Multirotor by using keys.
Anyway if someone figures out what should be done to get it work, please let others know.

Thanks,
Huyen

sytelus, 2017-10-26:
I think I just found the fix for this issue and have pushed it. Please give it a try. You will need to build from source to use this fix. I'm planning to update the binaries tomorrow.

HuyenPhuongVu, 2017-10-27:
Hi, can you please tell more details what you have done to get it work?
Thanks lot :)

GitWyd, 2017-11-08:
@sytelus Yes, please let us know. I have had the same issue, but need to be able to fix it within my own fork.

sytelus, 2017-11-14:
Here's the commit for the fix: https://github.com/Microsoft/AirSim/commit/c02dcfd3175647c26dbc5388f5bd8e5be74f22f8

closed 2017-11-14 by sytelus