microsoft/AirSim #548: Build failed: UAirBlueprintLib is not a class or namespace name

zungam, 2017-10-17:
Hi, im trying to build the plugin. I launched build.cmd in Devloper command prompt for VS 2017. Everything seemed to work but it suddenly stopped. Ive tried to figure this out, but I dont understand what happened.

Error code
`Done Building Project "C:\Users\Ascend\Documents\Simulator\AirSim\AirSim.sln" (default targets) -- FAILED.

Build FAILED.

"C:\Users\Ascend\Documents\Simulator\AirSim\AirSim.sln" (default target) (1) ->
"C:\Users\Ascend\Documents\Simulator\AirSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
"C:\Users\Ascend\Documents\Simulator\AirSim\AirLib\AirLib.vcxproj" (default target) (3) ->
(ClCompile target) ->
  c:\users\ascend\documents\simulator\airsim\airlib\include\vehicles\multirotor\api\droneapi.hpp(269): error C2653: 'UA
irBlueprintLib': is not a class or namespace name [C:\Users\Ascend\Documents\Simulator\AirSim\AirLib\AirLib.vcxproj]
  c:\users\ascend\documents\simulator\airsim\airlib\include\vehicles\multirotor\api\droneapi.hpp(269): error C3861: 'Ru
nCommandOnGameThread': identifier not found [C:\Users\Ascend\Documents\Simulator\AirSim\AirLib\AirLib.vcxproj]

    0 Warning(s)
    2 Error(s)

Time Elapsed 00:01:23.42
#### Build failed`

Appriciate all help I can get

zungam, 2017-10-18:
**SOLVED: ** I tried to build on Cmake version "Release Candidate (3.10.0-rc2)". Dont do that! I uninstalled and used the cmake version latest realese "Latest Release (3.9.4)". It still didnt work in using the "VS2015 x64 Native Tools Command Prompt" as specified in the documentation, but switching to a new console "Developer command propmt for VS2017" and running with the stable cmake version did it.

closed 2017-10-18 by zungam

Arlen0615, 2017-11-02:
Hi zungam,

May I know how you use Latest Release (3.9.4) in "Developer command propmt for VS2017"?

I install cmake-3.9.4-win64-x64.msi done, but when I run build.cmd in AirSim/buid.cmd in Developer command propmt for VS2017, this script still download cmake-3.7.2-win64-x64.zip automatically
The script show Fetching: https://cmake.org/files/v3.7/cmake-3.7.2-win64-x64.zip

Thanks!

Arlen0615, 2017-11-02:
Sorry, I got it, when install cmake, you have to choose "Add CMake to the system path for all users"

but I still get following error with Developer command propmt for VS2017 + cmake 3.9.4 :
CMake Error at CMakeLists.txt:2 (project):
  Failed to run MSBuild command:
    MSBuild.exe
  to get the value of VCTargetsPath: