microsoft/AirSim #549: Difficulty hooking up an offboard computer while in HIL sim in AirSim

realdealneil, 2017-10-17:
I have a computer vision algorithm running on a companion computer that communicates with a Pixhawk autopilot and steers it around.  I am trying to now use AirSim to simulate missions.  I can run AirSim in a couple of different configurations: 
1) AirSim owns serial port.  I fly vehicle around from RC controller.  (This works)
2) AirSim owns serial port.  I fly around and forward telemetry data over to QGroundControl over UDP by editing AirSim settings.json file.  (This works)

However, if any other app owns the serial port, I can't seem to get AirSim working.  For example, if QGC owns the serial port, and I create a new connection in the Comm Links menu in QGC, I only seem to be able to receive data in QGC, not retransmit it.  When I then launch AirSim, it says to restart SIL or HIL because it's not getting any data.  

Most of my confusion is in how I would set up the connection between my external computer (which happens to be an Nvidia Jetson TK1 computer, running Ubuntu 14.04) and AirSim.  I see that there are a couple of different connection types in the settings.json file, such as "UdpIp" and UdpPort" (which I believe are used for SITL sim) and "QgcHostIp" and "QgcPort" which are for broadcasting telemetry from AirSim to QGC.  However, I was hoping there would also be an additional connection for a companion computer such that I could connect it up as well.  Am I misunderstanding how the different connections are supposed to work?  Is a bidirectional link between the companion computer and AirSim possible?  I'm fine with editing the code if this type of thing hasn't been supported yet, but I don't want to go there if it's already been done before and I'm just not cluing in on how the parameters work.

sytelus, 2017-10-19:
I've done bunch of these stuff sometime ago so my memory is bit hazy but here is some overview. AirSim uses MavLinkCom component developed by @lovettchris. The MavLinkCom has proxy architecture where by you can open connection to PX4 either using serial or UDP and then other components shares this connection. So basically when PX4 sends MavLink message, all components receives that message. If any component sends message then its received by PX4 only. This allows you to connect any number of components to PX4.

The code you want to look at is [this](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/controllers/MavLinkDroneController.hpp#L793). Here we open connection for LogViewer and QGC. You can add something more if you like.

If you want to use QGC + AirSim togather than you will need QGC to let own the serial port. QGC opens up TCP connection that acts as proxy so any other component can connect to QGC and send MavLinkMessage to QGC and then QGC forwards that message to PX4. So you tell AirSim to connect to QGC and let QGC own serial port.

For companion board, the way we did it earlier was to have Gigabyte Brix on the drone. This x86 full fledged computer that will connect to PX4 through USB. We had Ubuntu on Brix and ran [DroneServer](https://github.com/Microsoft/AirSim/tree/master/DroneServer). The DroneServer created API end point that we can talk to via C++ client code (or Python code) and it translated API calls to MavLink messages. That way you can write your code against same API, test it in simulator and then ran same code on actual vehicle. So the companion computer has DroneServer running along with client code. Please note that we have been iterating very heavily since past few months and haven't got around to test above set up again so you are in bit of self-serve mode. We would be glad to have any contributions in case you find any bugs/enhancements.

realdealneil, 2017-10-31:
Thanks for the information , Shital.  I will be getting back to this project next week and I'll try what you mention here.  I think I will first try hijacking the LogViewer connection.  If that works, I may try adding an additional connection as well.

realdealneil, 2017-11-11:
I ended up going at this a different direction by having a second computer control the vehicle over mavros using the telem2 port.  This made the extra TCP/UDP port unnecessary.  Thanks for your suggestions though.  They will come in handy when trying to use SITL instead.

closed 2017-11-14 by sytelus