microsoft/AirSim #550: 2 agents running at the same time

cangokalp, 2017-10-18:
Does the simulation allows/supports 2 quadcopter running in the environment at the same time?

sytelus, 2017-10-19:
Yes, it does but in recent refactoring we broke couple of things. They are easy to fix if you are willing to look in the code. We'll get around fixing it in 1-2 months.

spencerh-b, 2017-10-23:
@sytelus can you please notify us when this has been officially fixed?

cangokalp, 2018-02-15:
Is this available now?

DavidLSmyth, 2018-02-15:
Works for me, with settings.json: 
```
{
  "SettingdVersion": 1.0,
  "SimMode": "Multirotor",
  "DefaultVehicleConfig": "SimpleFlight",

  "SimpleFlight_0": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Disarmed",
    "ApiServerPort": 41451
  },
  "SimpleFlight_1": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Disarmed",
    "ApiServerPort": 41452
  }
}
```
Pass in the respective ports to MultiRotorClient.py on initialisation. Seems to be a problem where both drones attempt to go to the same home position on takeoff, can be fixed in the c++ code.

cangokalp, 2018-02-15:
@DavidLSmyth are you able to do the same thing with PX4? With simple flight I was able to get it to work and control both of them, but I can't make it work with PX4 - airsim says failed to connect - could not detected a connected px4, but I actually have px4 waiting the simulator.

DavidLSmyth, 2018-02-15:
No, I haven't tried this with PX4. From your other issue, you can set your vehicle config name in the multirotor blueprint, however I don't think vehicle config name is used any more. I have been using an older branch of AirSim but maybe starting off looking at [this commit](https://github.com/Microsoft/AirSim/commit/2d9c52c96a1b12c65626c3e6c32c409def60d168) might help

closed 2020-09-04 by jonyMarino