microsoft/AirSim #551: Is rendering necessary?

cangokalp, 2017-10-18:
Is there a way to get the data from the simulation without rendering?

sytelus, 2017-10-19:
Which data? Car or multirotor? For multirotor, you can do simulation without Unreal Engine. In that case you get all the kinematics data (pose, velocity, acceleration) but no images. Here is some examples of how to use AirSim backend without any rendering engine (see StandAlone* files):

https://github.com/Microsoft/AirSim/tree/master/Examples

cangokalp, 2017-10-25:
For multirotor, yes;

This way it'll behave exactly as how it would behave with the rendering, right? I mean would the physics be there? Also can I do this via python API?

sytelus, 2017-11-14:
Currently API server runs in Unreal code so you won't be able to use APIs. Related: https://github.com/Microsoft/AirSim/issues/563

closed 2017-11-14 by sytelus