microsoft/AirSim #553: Does reset work now?

cangokalp, 2017-10-18:
I pulled the repo again just now, but reset still won't work.
msgpackrpc.error.RPCError: b"rpclib: server could not find function 'reset' with argument count 0."

sytelus, 2017-10-19:
yes, it is implemented now and I've tested for car as well as multirotor. Could you please make sure you have latest AirSimClient.py? Above error message means you have older Python client.

cangokalp, 2017-10-20:
now the reset() works and the multirotor resets from the starting point. However, after the reset I can't control it anymore, it doesn't do anything -say when I do client.moveToZ(-10,5), it just sits there.

sytelus, 2017-10-21:
Are you using PX4 or simple_flight (default)?

zamlz, 2017-10-23:
Hi I just pulled today, but I still get the same error. Do I have to recompile my environment for it to work properly?

zamlz, 2017-10-24:
Alright, it looks like if I compile the unreal project, it works. Any news on when the binaries will be updated. I would really like to use the downtown and neighborhood environment with the reset feature.

cangokalp, 2017-10-25:
I'm using simple flight @sytelus

cangokalp, 2017-10-26:
I'm using it in the blocks environment. When I do reset, it resets but I can't control it after that.

sytelus, 2017-10-26:
Ok... I found the issue. There was a bug that I just pushed the [fix](https://github.com/Microsoft/AirSim/commit/c02dcfd3175647c26dbc5388f5bd8e5be74f22f8). However another thing is that you must call below again after reset() because we reset the complete state.

```
client.enableApiControl(True)
client.armDisarm(True)
```

Below is the sample code I tested:

```
from AirSimClient import *

# connect to the AirSim simulator 
client = MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

AirSimClientBase.wait_key('Press any key to takeoff')
client.takeoff()

AirSimClientBase.wait_key('Press any key to move vehicle to (-10, 10, -10) at 5 m/s')
client.moveToPosition(-10, 10, -10, 5)

AirSimClientBase.wait_key('Press any key to reset to original state')
client.reset()
client.enableApiControl(True)
client.armDisarm(True)

AirSimClientBase.wait_key('Press any key to takeoff')
client.takeoff()

AirSimClientBase.wait_key('Press any key to move vehicle to (-10, 10, -10) at 5 m/s')
client.moveToPosition(-10, 10, -10, 5)
```

closed 2017-11-02 by sytelus