microsoft/AirSim #554: Unable to Debug unreal project VS

JJ-L, 2017-10-19:
Just downloaded the AirSim repo and UE 4.16,
followed the step-by-step instructions for "Creating and Setting Up Unreal Environment",
and upon reaching step 9 (Press F5) to run, the debugging failed with errors.

The error output were as follows:
1>------ Build started: Project: LandscapeMountains3, Configuration: DebugGame_Editor x64 ------
1>  Creating makefile for LandscapeMountains3Editor (no existing makefile)
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Source\LandscapeMountains3\LandscapeMountains3.Build.cs : warning : Module constructors should take a ReadOnlyTargetRules argument (rather than a TargetInfo argument) and pass it to the base class constructor from 4.15 onwards. Please update the method signature.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Source\LandscapeMountains3.Target.cs : warning : SetupBinaries() is deprecated in the 4.16 release. From the constructor in your .target.cs file, use ExtraModuleNames.Add("Foo") to add modules to your target, or set LaunchModuleName = "Foo" to override the name of the launch module for program targets.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Source\LandscapeMountains3Editor.Target.cs : warning : SetupBinaries() is deprecated in the 4.16 release. From the constructor in your .target.cs file, use ExtraModuleNames.Add("Foo") to add modules to your target, or set LaunchModuleName = "Foo" to override the name of the launch module for program targets.
1>  Performing full C++ include scan (no include cache file)
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\bench\basicbenchmark.cpp(1): error : Expected basicbenchmark.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\bench\benchmark.cpp(1): error : Expected benchmark.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\bench\btl\libs\BLAS\main.cpp(1): error : Expected main.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\bench\btl\libs\blaze\main.cpp(1): error : Expected main.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\bench\btl\libs\gmm\main.cpp(1): error : Expected main.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\bench\btl\libs\mtl4\main.cpp(1): error : Expected main.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\bench\btl\libs\STL\main.cpp(1): error : Expected main.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\bench\btl\libs\tvmet\main.cpp(1): error : Expected main.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\bench\btl\libs\ublas\main.cpp(1): error : Expected main.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\unsupported\doc\examples\MatrixExponential.cpp(1): error : Expected MatrixExponential.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\unsupported\doc\examples\MatrixFunction.cpp(1): error : Expected MatrixFunction.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\unsupported\doc\examples\MatrixLogarithm.cpp(1): error : Expected MatrixLogarithm.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\unsupported\doc\examples\MatrixPower.cpp(1): error : Expected MatrixPower.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\unsupported\doc\examples\MatrixSquareRoot.cpp(1): error : Expected MatrixSquareRoot.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\unsupported\test\EulerAngles.cpp(1): error : Expected EulerAngles.h to be first header included.
1>C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\Plugins\AirSim\Source\AirLib\deps\eigen3\unsupported\test\NumericalDiff.cpp(1): error : Expected NumericalDiff.h to be first header included.
1>  Build canceled.
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets(41,5): error MSB3073: The command ""C:\Program Files\Epic Games\UE_4.16\Engine\Build\BatchFiles\Build.bat" LandscapeMountains3Editor Win64 DebugGame "C:\Users\TVAGAME1\Documents\Unreal Projects\LandscapeMountains_jj - Car\LandscapeMountains3.uproject" -waitmutex" exited with code 1.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

What could be the cause of these error, and how should I rectify it?

closed 2017-10-23 by JJ-L

Arlen0615, 2017-11-04:
I got totally same issue with JJ-L, do anyone know what should be install?

ak4728, 2017-11-21:
Anybody knows how to fix this?

kalpitsmehta, 2018-02-04:
@JJ-L , @Arlen0615 Were you able to solve this error?

wujiyoung, 2018-05-07:
I find that you, @Arlen0615, sovled this problem in [Issue587](https://github.com/Microsoft/AirSim/issues/587) by updating your system to win 10. I can only work on win 7, so I wonder is there any solution on win 7?