microsoft/AirSim #560: Easiest way to access vehicles_ vector for multirotor api?

spencerh-b, 2017-10-23:
I have debugged through the Unreal and Found the vehicles_ vector with in the simmodeworldbase.h file in which all of the vehicle pawns ptrs are located, but before I make a mess of things I figured I would ask the easiest way to access the vector in order to control individual pawns through an api.

spencerh-b, 2017-10-26:
I managed to figure it out for better or worse by changing the files 
SimModeWorldMultiRotor.h/.cpp by making all the drones set up on a different port which was apparently not being changed via the .json file.

closed 2017-10-26 by spencerh-b

saihv, 2017-12-09:
Does control work fine with those modifications? (with PythonClient or DroneShell). I have been looking into making similar modifications until a fix is released.

spencerh-b, 2017-12-09:
Yeah it really only takes a couple lines of code. Specifically, the bottom function in SimModeWorldMultiRotor.cpp incrementing the port by utilizing the size of the vehicles vector.

saihv, 2017-12-09:
Weird. I tried the exact same thing, and the API port changes, but the python client doesn't connect to any other port apart from 41451.

saihv, 2017-12-09:
My code is at https://github.com/Microsoft/AirSim/issues/664#issuecomment-350501834

spencerh-b, 2017-12-09:
Ohhhh I forgot I as well added into the python client. My apologies. Here is a copy of my AirSimClient
[AirSimClient.py.zip](https://github.com/Microsoft/AirSim/files/1545078/AirSimClient.py.zip)

saihv, 2017-12-09:
Thanks for the link. Is this the only change you made? Because I have the same change, yet it doesn't work.

````
def __init__(self, ip, port):
        self.client = msgpackrpc.Client(msgpackrpc.Address(ip, port), timeout = 3600)
````

spencerh-b, 2017-12-09:
I added a port variable into three functions I believe and then you have to input it when you initialize the drone in your api

spencerh-b, 2017-12-09:
`class Airsim_Swarm:
    def __init__(self):
        #########################################
        #Number of Swarm Agents
        self.num_clients = 3
        self.num_cameras = 2
        #########################################
        '''
        Create list of client drones
        '''
        self.clients = []
        for i in range(0, self.num_clients):
            self.clients.append(MultirotorClient(41451+i))
            self.clients[i].confirmConnection()
            self.clients[i].enableApiControl(True)
            self.clients[i].armDisarm(True)`

saihv, 2017-12-09:
I am doing pretty much the same thing. As soon as I try to connect on 41452 or any other ports, I get 

`WARNING:tornado.general:Connect error on fd 440: WSAECONNREFUSED`

It's confusing. Thanks for the help anyway.

spencerh-b, 2017-12-09:
@saihv I have never seen an error like that, try using 41453 to check if that port is maybe being used by a different process

saihv, 2017-12-09:
Looks like it's not even because of the port number. Are you using the latest version? Or an older commit?

spencerh-b, 2017-12-10:
@saihv Older commit because I am on ubuntu 14.04 otherwise it wont build a62163c5c68eb066c742326828cd7c6159948712

saihv, 2017-12-24:
Weirdly, I still have the same issue with that commit. Irrespective of the port numbers, only the first vehicle responds to the API control request, none of the others do. I think I have the same modification as yours in `SimModeWorldMultiRotor.cpp`

`vehicle_params_.push_back(std::move(vehicle_params));

    std::shared_ptr<MultiRotorConnector> vehicle = std::make_shared<MultiRotorConnector>(
        wrapper, vehicle_params_.back().get(), enable_rpc, api_server_address,
        41454 + vehicle_params_.size() - 1, manual_pose_controller);
`

saihv, 2017-12-24:
I think I figured out what's wrong: don't you also need to do vehicle->startApiServer() for each of the vehicles? Which means your fpv_vehicle_connector should be an array instead of just one

spencerh-b, 2018-01-30:
Sorry for the late response I took a little vacation time. Interesting I have been able to have this running and I'm not really sure what the issue is. I have also been able to get this running with python, but it seems as though a fix that utilizes the .json file is necessary for total reliability