microsoft/AirSim #561: Issue with recording settings

oppne, 2017-10-24:
I set my recording settings
  },
  "CaptureSettings": [
    {
      "ImageType": 0,
      "Width": 2048,
      "Height": 2048,
      "FOV_Degrees": 90,
      "AutoExposureSpeed": 100,
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    }
  ],

Doesnt really change anything, no matter what settings i change in the settings json, nothing changes in ue4. Recordings always standard settings.

oppne, 2017-10-30:
Still not working, anyone have a solution?

oppne, 2017-10-30:
I fixed it.

Now how can i render out different image types at the same time?
TRied adding to array, didint work properly.

sytelus, 2017-11-14:
Please see this and continue conversation there: https://github.com/Microsoft/AirSim/issues/607

closed 2017-11-14 by sytelus

Beijingsheng, 2018-08-09:
@oppne  How do you fix your problem? I'm facing the same problem right now. No matter what I write in settings.json won't cause any changes.