microsoft/AirSim #562: How can I change the default height achieved by the drone while taking off the drone

shristi945, 2017-10-24:
Since I am new to airsim I am curious to now where I can set my desired height attained while taking off the drone as I have gone through the code in DroneBaseController.cpp the takeoff function is calling getTakeoffZ() function and the function in its definition returns  params_.takeoff.takeoff_z  and returning a value say -2.0f is not working.
If anyone can help than please reply.
Thanks.

Honky103, 2017-10-25:
You can set it from QGroundControl, when you set its takeoff waypoint, you can set its takeoff altitude.

shristi945, 2017-10-25:
I am not using QGroundControl for now. I am just using simple flight.

spencerh-b, 2017-10-25:
I am a little confused by your question. Are you wondering how you can take off to a certain height? Or are you wondering how you can obtain the height you are at after taking off?

shristi945, 2017-10-25:
@spencerh-b  I am wondering about how can I obtain a desired height after taking off. I mean after taking off drone attains certain default height (min 3m upwards). I want to change the value of height attained by the drone after taking off.

spencerh-b, 2017-10-25:
@shristi945 I wouldn't suggest changing the takeoff function with in the client profile, but rather simply using the MoveToZ function, in which, you can input your desired height and speed at which you would like the vehicle to move. Alas, if you are deadset on changing the takeoff function itself you will have an understanding of Unreal Cpp classes and clients and you will want to explore within the msgpack of the rpclib which is located here:

`AirSim\AirLib\deps\rpclib\include\rpc\msgpack`

shristi945, 2017-10-26:
@spencerh-b  considering your suggestion I am using MoveToZ function to do so for now and it works Thanks.

closed 2017-10-26 by shristi945