microsoft/AirSim #563: multiple airsim simulations at the same time

cangokalp, 2017-10-25:
Is there a way to run more than one simulation at a time? that way I can test different models running in different simulations

cangokalp, 2017-11-02:
Ideally, I want to be able to not render the environment (I just need the physics) and have multiple instances of the simulation so that I can parallelize the RL algorithm or maybe run different models on different instances for comparison of models.

sytelus, 2017-11-14:
I think you don't want render but you still want all objects in environment and collision detection. Unfortunately Unreal doesn't allow turning off rendering. It might be possible (its open source engine) but I don't know... But in any case, if you turn off rendering then you won't have ability to get screenshots or depth view etc.

If you just want physics without collisions and any other objects in environment then its doable. Here is some sample code: https://github.com/Microsoft/AirSim/blob/master/Examples/StandAlonePhysics.hpp

closed 2017-11-14 by sytelus

maximilianchang, 2017-12-01:
@sytelus, @cangokalp  : could you elaborate on this? I want to run A3C and I do not follow how to go from this thread to being able to parallelize the RL algorithm. 

Let's say I want to train an A3C Algorithm to train a car how to drive in the neighborhood environment (so physics and collision info is relevant).  How would I go about doing this?

I have looked at the following:
1. StandAlonePhysics.hpp example as included in this thread. 
2. Documentation on Flying Multiple Objects in the same environment (multi_vehicle.md) 
but don't think any of these resources adequately address this issue. I imagine it has something to do with configuring ports, but it is unclear how to do this with the given APIs.

mallela, 2018-11-12:
I want to do something similar to what @maximilianchang suggests. I need the physics and collision information. 

Is there a way to have multiple instances of the same UE project running on different ports and control the respective clients by referencing them uniquely in the PythonClient?

This will also be useful in training different models in parallel. 
@sytelus @lovettchris

AloshkaD, 2018-12-23:
I've been trying to do the same but I wasn't successful.

PHRABAL, 2018-12-29:
Training a drone on a RDL algo where you run one thread in real time takes forever. It can take several million time steps to properly train an agent depending on the task. At a couple time steps per second (real time) and 4 million time steps needed, as an example, that's 23 days to train it. What are people doing to speed things up? Many Unity ML agent environments provide for multi-threads and, in games of course each time steps happens much faster.

AloshkaD, 2018-12-30:
@PHRABAL you are absolutely right. It takes ages to train RL with a single env and the current approach does not scale. @sytelus we appreciate any update you may have on supporting multi-envs.

zsfVishnu, 2019-01-18:
@PHRABAL ,did the multi-threading work for you in Unity for AirSim?

PHRABAL, 2019-01-18:
No. As far as I know, there is no multi-threading with AirSim, so I am training one drone in the environment.

mallela, 2019-01-18:
If you want to train multiple agents in parallel or run multiple simulations at a time, you can use different local host IPs to do it.

1. First you need to add the following to your AirSim's settings.json file:
 `"LocalHostIp": 127.0.0.x` 
Here, replace x with your choice of number. The default is 127.0.0.0

2. When you launch your environment, this new IP is used. 
3. Next, you have to pass this same IP to your AirSim car client in python:
`client = airsim.CarClient(ip=127.0.0.1)`

If you want to train another agent, simply change the LocalHostIp in the settings.json file, launch another AirSim instance, and pass the new IP in the PythonClient code.

zsfVishnu, 2019-01-21:
Yea, multithreading is possible via changing thr LocalhostIp. Just tried it with Unreal Engine.

PHRABAL, 2019-01-24:
Yes, it functions, but it slows down the simulation a lot, too much. I'm running on a decent g3 aws instance and when I run two sims at the same time it goes from ~ 60 milliseconds per cycle to sometimes over 200 milliseconds. I need to have the drone's reaction time be quick and also consistent from one time step to the next, so this solution doesn't work. The only thing I have come up with is simply to run multiple separate aws instances, one for each sim I want to run.