microsoft/AirSim #567: the simulation often crashes

cangokalp, 2017-10-26:
which version of unreal engine is more stable, currently my simulation often crashes sometimes after thousand training episodes sometimes sooner, sometimes I get this error:
Exception thrown at 0x00007FFCE379B540 (UE4Editor-Core.dll) in UE4Editor.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF.
and sometimes it just says Blocks stopped working.

sytelus, 2017-10-26:
I've started noticing one crash scenario and traced to potential issue in rpclib so I've filed [issue there](https://github.com/rpclib/rpclib/issues/161). This is when client sends command but disconnected before command finishes execution. However I'm not sure if your issue is same. I would suggest to try to create some minimal scenario with AirSim so crash can be reproduce. If you run from source code that would help as well to see the call stack and location of the crash.

cangokalp, 2017-10-27:
When it crashes I just get a windows error window that says Blocks stopped working and then if I hit debug I see;

Unhandled exception thrown: read access violation.
this->values_._Mypair._Myval2._Myhead->_Next was 0x100000050.
If there is a handler for this exception, the program may be safely continued.

```
    virtual void update() override
    {
        UpdatableObject::update();

        if (!times_.empty() && 
            ClockBase::elapsedBetween(clock()->nowNanos(), times_.front()) >= delay_) {

            last_value_ = values_.front();
            last_time_ = times_.front();

            times_.pop_front();
            values_.pop_front();
        }
    }

```

cangokalp, 2017-10-29:
I was suspicious about the blocks environment, so I created a new very basic environment - but still got the error:
Unhandled exception at 0x00007FF7850161AE in Quadcopter.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF

What's causing this? Error changes from time to time but it's similar to this usually

cangokalp, 2017-10-29:
I have one more input - I think this issue is somehow related to the collisions. If I reset before collusion occurs my training keeps going without simulation crashing. However this way I can't achieve what I want to do - temporarily I do this so that the training keeps going.

sytelus, 2017-11-14:
What are the repro steps?

Kjell-K, 2017-11-23:
@sytelus I think we are still taking about the same issue  #631.
Any solutions yet? This issue stands in the way of performing long time simulations for reinforcement learning. Note that I never had this issues with the AirSim version from early October. Here I was running simulations for 4 days.

sytelus, 2017-12-01:
I think latest version should fix this. Lets continue discussion here anyway: https://github.com/Microsoft/AirSim/issues/631

closed 2017-12-01 by sytelus