microsoft/AirSim #569: Block environment : ImageWrapper.h file not found

jeremyfix, 2017-10-27:
Hi, when loading the Block UE environment you provide, UEditor is compiling it and at some raises an issue :

In file included from /opt/GIT/AirSim/Unreal/Environments/Blocks 4.18/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:14:
/opt/GIT/AirSim/Unreal/Environments/Blocks 4.18/Plugins/AirSim/Source/AirSimGameMode.cpp:3:10: fatal error: 
      'ImageWrapper.h' file not found
#include "ImageWrapper.h"

I tried to "find . -name  ImageWrapper.h" but nothing is returned neither in AirSim nor in UnrealEngine. What is this missing file coming from ?

jeremyfix, 2017-10-27:
A follow up of my question. I'm using UnrealEngine **4.18** ; I found that the header where the IImageWrapperModule is defined is actually UnrealEngine/Engine/Source/Runtime/ImageWrapper/Public/IImageWrapperModule.h . There is also a IImageWrapper.h header. 

I might give a try changing the include. Do you agree on this change ?

Fred12138, 2017-10-28:
I meet the same question as you. How about you now?

jeremyfix, 2017-10-28:
I actually succeeded this morning to compile the plugin by changing the hearder 

In AirSim/Unreal/Plugins/AirSim/Source/AirSimGameNode.cpp , change

#include "ImageWrapper.h" 

by

#include "IImageWrapperModule.h"

Then adding the plugin to a new project, it compiled and I was able to spawn a drone.

jeremyfix, 2017-10-28:
I created a pull request #570

Fred12138, 2017-10-28:
Yeah I tried. your can do it actually!

sytelus, 2017-11-02:
Thanks for figuring this out. Your PR is now merged!

closed 2017-11-02 by sytelus