microsoft/AirSim #571: VR Build

eanmikale, 2017-10-29:
We would like to test prototypes and build environments with the VR Editor Mode in Unreal, utilizing the HTC Vive and/or the HP Mixed Reality Headset. Is there any support coming soon for this capability? Thank you.

sytelus, 2017-11-02:
Yes, this is in plan. Its fairly easy to add with Unreal Engine so please feel free to look in to it. We might not get around implementing this for at least next 2 months.

eanmikale, 2017-11-10:
We have it built. It is quite the experience. Recommendations for future builds. We'd love to be able to utilize the command prompt, in-game for complete immersion. Thanks again.

closed 2017-11-10 by eanmikale

kandai-wata, 2018-01-04:
Could you tell us how you managed to build it for VR? 
I am trying to view it with oculus rift.