microsoft/AirSim #572: 'msr::airlib::RealMultirotorConnector' is an abstract class

jeremyfix, 2017-10-30:
When running   build.sh , an error is thrown because 'msr::airlib::RealMultirotorConnector' is an abstract class.   Indeed,  the following two virtual pure methods are not overriden :

virtual bool setSegmentationObjectID(const std::string& mesh_name, int object_id,
				       bool is_name_regex = false) = 0;
  
  virtual int getSegmentationObjectID(const std::string& mesh_name) = 0;

in the file AirSim/AirLib/include/vehicles/multirotor/controllers/RealMultirotorConnector.hpp  and it inherits from AirSim/AirLib/include/controllers/VehicleConnectorBase.hpp

nwayt001, 2017-11-01:
I had the same error, did what you did and it worked for me. I'm not sure what effect this will have though.

sytelus, 2017-11-02:
Thanks for reporting this. This is now [fixed](https://github.com/Microsoft/AirSim/commit/c85eb438a7b2ad795a75f10f9deccafa447293de).

closed 2017-11-02 by sytelus

tmericli, 2017-11-28:
The problem still seems to persist; this time complaining about the following virtual function that is not implemented in RealMultirotorConnector.hpp:

`virtual void printLogMessage(const std::string& message, std::string message_param = "", unsigned char severity = 0)`

Temporarily adding that function to RealMultirotorConnector.hpp with empty body solves the build problem, but a more principled solution is needed.