microsoft/AirSim #574: RC Controller on USB not detedcted

JeongsooHa, 2017-10-30:
OS : Linux(ubuntu)
I have installed 4.16.3 version of Unreal Engine and 66f9635 commit number of AirSim.
I wanted to use remote controller(FS-TH9X) in simulator. So, I also have installed driver of pixhawk4 and Qgroundcontrol. Then, i connected RC with pixhawk4. I could see RC could control Qgroundcontrol. 
when i try to control simulator drone in Blocks project, but It have never worked well.
ERROR
RC Controller on USB not detected  : 18446744073709551615
[image]

zergler, 2017-11-01:
Did you set up the right [configuration](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md)? It could also be that QGroundControl is taking over the connection. Try running it without QGroundControl.

JeongsooHa, 2017-11-01:
Hi zergler.
Thank you for answer me. Unfortunately I already checked "settings.json". 
This is my settings.json file.
{ "SettingsVersion": 1, "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md"}
I think it is the right configuration. 
I checked the list of USB using "lsusb"
[image]
But I could not see the name of USB(26ac:0011)
(port number "26ac:0011" is for PX4 FMU v2.x)
[image]

So, I am confused now.
I don't know whether the USB(PX4) isn't working well or AirSim can not be getting the PX4 USB ID.
Thanks.

JeongsooHa, 2017-11-02:
This is SimJoyStick.cpp
[image]
(I am changing this code for checking variable)
 As you can see, devicePath is "/dev/input/js" << index. 

[image]
However, there is no js* file in "/dev/input/". just event*.

And This is my settings.json

{ "SettingsVersion": 1.0,
"SimpleFlight": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Armed",
    "RC": {
      "RemoteControlID": 14,
      "AllowAPIWhenDisconnected": false,
      "AllowAPIAlways": true
    },
    "ApiServerPort": 41451
  },
"PX4": {
    "FirmwareName": "PX4",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "VehicleCompID": 1,
    "VehicleSysID": 135,
    "ApiServerPort": 41451
  },
"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md"
}

sytelus, 2017-11-14:
AirSim currently only detects those joysticks which appears in /dev/input/js. Please see this which might be relevant: http://forum.vectorlinux.com/index.php?topic=18087.0. Please feel free to contribute the enhancement so we can detect wider array of joysticks!

closed 2017-11-14 by sytelus

JeongsooHa, 2017-11-14:
Hi sytelus,
Thank you for your reply. I think there will be a lot of help.