microsoft/AirSim #576: Include paths build error and Not located!

oppne, 2017-10-31:
Get these when i tried to build
`Severity	Code	Description	Project	File	Line	Suppression State
Error		File 'D:\Marius\AirSim\AirSim UE4 directory\TestingSimulator\Plugins\AirSim\Source\AirBlueprintLib.cpp' was included by multiple modules, but with different include paths	TestingSimulator	D:\Marius\AirSim\AirSim UE4 directory\TestingSimulator\Intermediate\ProjectFiles\EXEC	1	
Error		The first include statement in source file 'D:\Marius\AirSim\AirSim UE4 directory\TestingSimulator\Plugins\AirSim\Source\AirLib\src\safety\ObstacleMap.cpp' is trying to include the file 'thread' as the precompiled header, but that file could not be located in any of the module's include search paths.	TestingSimulator	D:\Marius\AirSim\AirSim UE4 directory\TestingSimulator\Intermediate\ProjectFiles\EXEC	1	
`

oppne, 2017-10-31:
I forgot to say that i added controlbase to modules, and added modulerules.build.cs in the source file of the plugin, just to see if i can make it to blueprints, maybe this is related?

sytelus, 2017-11-13:
Not sure what code changes were made but it all really depends on that. Also see https://github.com/Microsoft/AirSim/issues/588

closed 2017-11-13 by sytelus