microsoft/AirSim #578: How to run Neighbourhood or city downtown on Linux?

Arlen0615, 2017-11-01:
Hi, 

I have study **Setup Unreal Environment** part 

and I can run well **Option 1: Built-in Blocks Environment** on my Ubuntu PC

But I have no idea how to change the project to Neighbourhood or City downtown..

Is there any unreal project file on this github of Neighbourhood or City downtown?

Thanks very much.

zergler, 2017-11-01:
Adding AirSim to custom environments is explained well in this [doc](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md), go take a look. If you are referring to the binaries for the Neighborhood environment from the AirSim release, those are Windows binaries so they wouldn't run on Linux.

The basic procedure is to generate the project files using a Windows PC and then copy them over to your Linux setup. UE4 for Linux is missing the Epic Games Launcher, which is required for this.

mickey13, 2017-11-01:
Is there a (supported or unsupported) way to generate the appropriate project files on Mac OS, since it has the Epic Games Launcher?

zergler, 2017-11-01:
It would probably be the same general procedure on a Mac.

closed 2017-11-02 by sytelus