microsoft/AirSim #579: files VehicleCameraBase.hpp & Utils.hpp not found

ProjectVenom, 2017-11-01:
I get the following error when recompiling the block environment with AirSim on UE4.17.

```
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/PIPCamera.h:6:10: fatal error: 
      'controllers/VehicleCameraBase.hpp' file not found
#include "controllers/VehicleCameraBase.hpp"
         ^
1 error generated.
In file included from /home/projectvenom/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:2:
/home/projectvenom/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.cpp:11:10: fatal error: 
      'common/common_utils/Utils.hpp' file not found
#include "common/common_utils/Utils.hpp"
         ^
1 error generated.
ERROR: UBT ERROR: Failed to produce item: /home/projectvenom/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so

```

Have these files been moved?
Is there a fix in progress?

GitWyd, 2017-11-01:
Same problem here. Cloned AirSim, Compiled, copied Plugins Folder to Blocks environment, and tried to build it with UE4.17.

sytelus, 2017-11-02:
Could you please try UE 4.16. We are not yet supporting 4.17 yet. I've had a successful compilation with 4.17 however. Above error looks like you might have missed a build step. You might want to check following folders and see if these files exists:

AirSim/Unreal/Plugins - if you did build file should exist here
AirSim/Unreal/Environments/Blocks/Plugins - if executed update_from_git.bat then it copies above folder to this folder.

Hope this helps you to figure out which step file is missing

closed 2017-11-13 by sytelus