microsoft/AirSim #586: simSetPose exception with std::bad_cast

younghj, 2017-11-03:
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/home/tommy/code/flot/workspace/environment/AirSimControl.py", line 78, in run
    self.client.simSetPose(newPose, True)
  File "/home/tommy/code/flot/external/AirSim/PythonClient/AirSimClient.py", line 223, in simSetPose
    self.client.call('simSetPose', pose, ignore_collison)
  File "/usr/local/lib/python3.5/dist-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/usr/local/lib/python3.5/dist-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: b"rpclib: function 'simSetPose' (called with 2 arg(s)) threw an exception. The exception contained this information: std::bad_cast."

very simply when I try to run simSetPose to reset the drone, it doesn't work and this exception comes up.

sytelus, 2017-11-07:
What code are you executing? Very likely you are passing bad data structure in parameters.

younghj, 2017-11-07:
this problem has been solved
https://github.com/Microsoft/AirSim/issues/553

closed 2017-11-07 by younghj