microsoft/AirSim #588: MavLinkDroneController.hpp contains non-inlined definitions, resulting in linker errors

cdyk, 2017-11-04:
In the file

AirLib/include/vehicles/multirotor/controllers/MavLinkDroneController.hpp

from line 1280 an on, members of MavLinkDroneController are defined outside the class declaration, and is thus not automatically inlined.

This code will produce link symbols, so that if this header-file is included in more than one compile-unit, the linker (VS2017) will fail due to multiple definitions of the same symbol.

I guess the easiest solution is to move the pimpl-class and the forwarding functions into a cpp-file?

sytelus, 2017-11-04:
Arebyoy getting this on VS2017? I had tested the build and it seems to compile fine. This file is using header only model for C++ code.

cdyk, 2017-11-04:
Yes, using VS2017. However, the bundled examples are mostly a single compile unit, so those will not exhibit this behaviour.

I have a simple project in https://github.com/cdyk/AirSim that has this problem (though as warnings since I added the force output link flag).

Keeping the pimpl in the top of the hpp-file would allow the members of MavLinkDroneController to be defined inside the declaration, but might be a bit ugly.

Some header-only libraries use a define to wrap function definitions, and if only one compile unit has this define, functions gets defined only in that compile unit. Another variant, that doesn't break existing code, could be to add a define that excludes the function definitions?

sytelus, 2017-11-07:
I'm not sure if I understand correctly... Above file already has include gard. Do you these warnings if you try to compile with VS2017 without making any changes to AirSim code?

cdyk, 2017-11-07:
The include guards make sure that the file isn't included multiple times within a translation unit (a cpp-file), so when the project consists of a single cpp-file there is no problem. It seems that all relevant examples in AirSim consists of a single cpp file.

However, if a project contains more than one cpp-file that includes that header, the non-inlined implementations from the header produces the same symbols in multiple object files, and when the object files are linked, the linker finds multiple instances of the same symbols (LNK2005 in VS).

AFAIK, functions within a class definition are automatically inlined (and doesn't produce symbols). However, function definitions outside class definitions do produce symbols.

I've added #ifndefs around function definitions that are outside class definitions and are in header files in cdyk/AirSim@d9fb6d4bbee602f968af8d3e85544dd04ef92f51 , and I define AIRSIM_NO_IMPL in all but one cpp-file. This fixes the link issues for my project with multiple cpp-files.

sytelus, 2017-11-07:
I see the issue. I'm thinking about just moving methods outside of class definition to inside. I'll add this in my backlog.

jonyMarino, 2020-11-06:
already solved

closed 2020-11-06 by jonyMarino