microsoft/AirSim #591: Setup error on Ubuntu 16.04

alice17, 2017-11-06:
I cloned the repository in my home folder, tried to execute setup.sh but keeps giving me this error:

```
CMake Error: Cannot open file for write: /auto/homes/av508/Documents/AirSim/llvm-build/projects/libcxx/include/CMakeFiles/libcxx-headers.dir/depend.make.tmp
CMake Error: : System Error: Permission denied
projects/libcxx/include/CMakeFiles/libcxx-headers.dir/build.make:70: recipe for target 'projects/libcxx/include/CMakeFiles/libcxx-headers.dir/depend' failed
make[3]: *** [projects/libcxx/include/CMakeFiles/libcxx-headers.dir/depend] Error 2
CMakeFiles/Makefile2:12495: recipe for target 'projects/libcxx/include/CMakeFiles/libcxx-headers.dir/all' failed
make[2]: *** [projects/libcxx/include/CMakeFiles/libcxx-headers.dir/all] Error 2
CMakeFiles/Makefile2:12588: recipe for target 'projects/libcxx/lib/CMakeFiles/install-libcxx.dir/rule' failed
make[1]: *** [projects/libcxx/lib/CMakeFiles/install-libcxx.dir/rule] Error 2
Makefile:3707: recipe for target 'install-libcxx' failed
make: *** [install-libcxx] Error 2

```

sytelus, 2017-11-07:
Your error message says `Permission denied` for `/auto/homes/av508/Documents/AirSim`. So you might want to look at why.

closed 2017-11-13 by sytelus

nalinraut, 2019-02-01:
try `sudo setup.sh`. Or edit the shell script with `sudo` privileges where ever necessary.