microsoft/AirSim #592: I cannot use python clients

billchenxi, 2017-11-06:
Can someone help me with this error?

`C:\Users\Bill\Documents\GitHub\AirSim\PythonClient>python hello_car.py                                                  WARNING:tornado.general:Connect error on fd 536: WSAECONNREFUSED                                                        WARNING:tornado.general:Connect error on fd 536: WSAECONNREFUSED                                                        WARNING:tornado.general:Connect error on fd 536: WSAECONNREFUSED                                                        WARNING:tornado.general:Connect error on fd 536: WSAECONNREFUSED                                                        WARNING:tornado.general:Connect error on fd 536: WSAECONNREFUSED                                                        Waiting for connection: Traceback (most recent call last):                                                                File "hello_car.py", line 5, in <module>                                                                                  client.confirmConnection()                                                                                            File "C:\Users\Bill\Documents\GitHub\AirSim\PythonClient\AirSimClient.py", line 158, in confirmConnection                 home = self.getHomeGeoPoint()                                                                                         File "C:\Users\Bill\Documents\GitHub\AirSim\PythonClient\AirSimClient.py", line 167, in getHomeGeoPoint                   return GeoPoint.from_msgpack(self.client.call('getHomeGeoPoint'))                                                     File "C:\Users\Bill\Miniconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call                                 return self.send_request(method, args).get()                                                                          File "C:\Users\Bill\Miniconda3\lib\site-packages\msgpackrpc\future.py", line 43, in get                                   raise self._error                                                                                                   msgpackrpc.error.TransportError: Retry connection over the limit`

Many thanks,

sytelus, 2017-11-07:
Are you running AirSim as well? Above error means your Python client code is not able to find AirSim server. This may also happen if you didn't allowed AirSim to open ports (Windows Firewall shows dialog box and you might have pressed Cancel or Escape).

closed 2017-11-13 by sytelus

maximilianchang, 2017-11-24:
I ran into a similar situation. I am just running the out-of-box environment from "Neighborhood" (as downloaded from the link provided. 

Could the second issue you stated still apply? (Windows Firewall shows dialog box). And if so, do you have any pointers on how to resolve this?

cseas, 2018-07-19:
Can someone please explain how to fix this issue? I'm getting the same and can't figure out a way to make the Python client work. I'm trying to run DQNcar.py with the Neighbourhood binary.

ctselas, 2019-01-22:
@sytelus Do you have any answer ? Why this issue was closed since there was no answer and the problem still occurs in other persons ?