microsoft/AirSim #593: Stable camera position for VR

spencerh-b, 2017-11-06:
I have been using air sim in VR and it can be quite dizzying if the camera director is in the fly with mode. What I want to do, is make the camera director only spin around the z axis so it does not induce this dizzying effect on a user. Is there any where to do this?

sytelus, 2017-11-07:
Wouldn't be better to use FPV view instead of Fly with me? Also try out new SpringArmChase mode (press "/" key).

spencerh-b, 2017-11-07:
@sytelus The issue with the fly with me view currently is the forward tilt of the drone. 

Example:
When moving forward in the drone the drone pitches forward. In this case, the users eyes are telling the user that they are about to experience a change in balance, which tell's there brain, lean back, or expect to feel some sort of shift in the forward direction, however, since this does not come there is a sort of vertigo effect induced on the user. 4 out of the 5 people so far who have tried this particular VR setup have had to stop from getting very dizzy. 

Currently the default camera director does a great job of slowly spinning around the drone so this is not to much of an issue when spinning, this is why I'm hoping to just eliminate the pitch and roll movements from the camera.

closed 2017-11-08 by spencerh-b