microsoft/AirSim #594: build error

abc19850508, 2017-11-07:
when I build the project,  I met the error 
"c:\users\windows\source\repos\airsim\airlib\include\common\earthutils.hpp(40): error C4579: 'msr::airlib::EarthUtils::D
ECLINATION_TABLE': in-class initialization for type 'const int [13][37]' is not yet implemented; static member will rem
ain uninitialized at runtime but use in constant-expressions is supported [C:\Users\Windows\Source\Repos\AirSim\AirLib\
AirLib.vcxproj]"
does anyone know how to solve this problem? thanks

sytelus, 2017-11-07:
Are you using Visual Studio 2015 Update 3? Looks like your C++ compiler is older.

abc19850508, 2017-11-07:
@sytelus I install visual studio 2015 with ISO on a new computer with win10 enterprise, but I can't install the Update 3 now which make me really upset(several days ago Update 3 could be installed and I build the airsim successfully, I found the reason might be that Microsoft dont want to support vs2015 any longer).  so which way should I try now? could AirSim be built with visual studio 2017 now? (and I cant install .net framework 3.5 on my computer system either, I dont know whether it would have any effect or not)

sytelus, 2017-11-07:
VS2015 Update 3 is supported by Microsot. You can download from: https://blogs.msdn.microsoft.com/visualstudio/2016/06/27/visual-studio-2015-update-3-and-net-core-1-0-available-now/,

We will be soon also update everything to VS2017 soon. The only reason we haven't done so is because this will force too many of our users to go through that process.

abc19850508, 2017-11-09:
@sytelus I cant update vs2015 with the vs2015.3.exe. Maybe the reason is the internet limit. I download  a visual studio 2015 with update 3 and repair vs2015 using it. Finally AirSim was built successfully. Thanks a lot for your help!

closed 2017-11-13 by sytelus