microsoft/AirSim #595: can AirSim be built with visual studio 2017?

abc19850508, 2017-11-07:
recently, i found I couldn't download the visual studio 2015 and the update ,can AirSim be built with visual studio 2017 now? and I cant install the .net framework 3.5, dose it have effect on the AirSim?

abc19850508, 2017-11-10:
not yet

sytelus, 2017-11-14:
I was just able to compile and run everything in VS2017.

Use "Developer command Prompt for VS 2017" to run build.cmd:

[image]

Select no upgrade for Platform toolset when opening .sln file:

[image]

That's it.

abc19850508, 2017-11-15:
@sytelus that‘s great！

Jazzinghen, 2017-11-16:
I actually had to upgrade the platform toolset in order to build the plugin.

sytelus, 2017-11-23:
@Jazzinghen which version of Windows are you using?

Jazzinghen, 2017-11-28:
I am using Windows 10 64-bit

n1ckfg, 2017-11-29:
You also have to edit build.cmd depending on whether you're using VS 2015 or VS 2017. Afaik if you have both on your system, you must build AirSim with VS 2017.

sytelus, 2017-12-01:
I have both on my system and I use VS2015 without any change in build.cmd. I'm also able to build VS2017 without any change in build. The only thing is that you have to use "Developer command Prompt for VS 2017".

Jazzinghen, 2017-12-04:
I have changed build.cmd, retargeted the library using Visual Studio and built from the Developer Command Prompt.
I should try your approach, maybe it would work out of the box.

sytelus, 2018-04-12:
We will be upgrading to VS2017 in short order.

closed 2018-04-12 by sytelus