microsoft/AirSim #598: 3 error in Simple flight Mode

winszenYoung, 2017-11-07:
My Computer Environment Specific
Windows 10
Unreal Engine 4.16.3
Airsim commit a4190ac68764cdfd08ca68cbaf29a840cfc184e3
Json Settings 
"DefaultVehicleConfig": "SimpleFlight",
"PhysicsEngineName": "FastPhysicsEngine",

1.  It is not work ignore_collision in simsetpose

olivia-skydio, 2018-01-31:
Any updates on this bug? I'm also facing this issue when trying to ignore collisions in computer vision mode.

cangokalp, 2018-02-23:
Yeah, and I'm facing the the simSetPose issue as well. Once this command is issued, issuing other commands have no affect.

saihv, 2018-06-01:
I have the same problem with the latest binaries.

jonyMarino, 2020-11-06:
Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing an issue
*Attach a screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-11-06 by jonyMarino