microsoft/AirSim #599: Using stationary external camera

spencerh-b, 2017-11-07:
Is there any way to place an external camera and use it as the default camera. I have 40 drones flying separately and want to view them from an above vantage point. At a minimum I would like them to record what they see.

sytelus, 2017-11-07:
You can use manual mode (press M). This causes camera to stop moving and you can control camera by arrow, page up/down and asdw keys.

spencerh-b, 2017-11-07:
Yes, but is there currently a way to place cameras through out an environment and to have them record from there stationary position?

spencerh-b, 2017-11-08:
Trying to set manual mode to the default in the settings.json triggers a break

spencerh-b, 2017-11-09:
@sytelus But is there a way to place multiple stationary external camera's within an environment and have them record when the environment is started up?

closed 2017-11-10 by spencerh-b

jmariscal-harana, 2020-01-28:
Hey, did you fix this in the end? Cheers!

spencerh-b, 2020-01-30:
@jmariscal-harana I ended up just programing extra drones to stay stationary; however, I would love to see a working solution to stationary cameras. I have since dropped this as a focal point, but could utilize this work in the future.

PR #3320 (2021-07-26): Add support for fixed external cameras

Fixes: #3208 
Fixes: #1896, 
Fixes: #1709, 
Fixes: #610, 
Fixes: #599
 

## About

This PR adds the capability to create a fixed camera separate from the cameras on the vehicle, such as a CCTV camera. I'm not aware of any simple way to go about doing this currently, and therefore thought to try it out. I haven't checked the issues very thoroughly, so please let me know if there are other related issues and ways to go about doing this.

PR has been opened to get ideas and suggestions and whether this is actually required. This will likely become a big PR, [... cut]