microsoft/AirSim #601: RPC Link Error when Packaging on Linux

kuanting, 2017-11-08:
Hi there,
I am trying to package Linux binary for users who do not have Unreal. However I encountered the following RPC link error:

UATHelper: Packaging (Linux): UnrealBuildTool: /data/UnrealProjects/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a(this_handler.cc.o): In function `rpc::this_handler()':
UATHelper: Packaging (Linux): UnrealBuildTool: 
 /home/user/Research/AirSim/external/rpclib/lib/rpc/this_handler.cc:10: undefined reference to `__cxa_thread_atexit'

I have successfully packaged binary on Windows, this issue only happens on Linux. Could you please show me how to add the link library to solve this issue? Thanks a lot.

sytelus, 2017-11-09:
This was traced to rpclib and then to a problem in Unreal's toolchain. Please see this: https://github.com/rpclib/rpclib/issues/140. Tamas's suggestion was to include below code in one of the cpp files in AirSim instead of rpclib. You can try that out and see if it works. If it does then please let us know (or send us pull request).

```
extern "C" int __cxa_thread_atexit(void (*func)(), void *obj,
                                   void *dso_symbol) {
  int __cxa_thread_atexit_impl(void (*)(), void *, void *);
  return __cxa_thread_atexit_impl(func, obj, dso_symbol);
}
```

Related thread of StackOverflow:

https://stackoverflow.com/questions/29322666/undefined-reference-to-cxa-thread-atexitcxxabi-when-compiling-with-libc/30437761#30437761

mitchellspryn, 2017-12-10:
I ran into the same thing. I tried the code fix suggested by @sytelus, but I could not get AirLib to compile. I added it to a few files (AirSim.cpp, CameraDirector.cpp, its own file), and I got the same linker error, only when compiling AirSim.

I found a solution. @kuanting, I assume that you already have a source build for UE4, as [cross compiling for linux from windows is only supported with source builds when using plugins](https://answers.unrealengine.com/questions/513922/code-plugin-prevents-packaging-for-linux-on-window.html). Also, I assume you've already gone through the steps [here](https://docs.unrealengine.com/latest/INT/Platforms/Linux/GettingStarted/index.html) to set up your environment. 

In this case, the easiest way to pass the needed linker flags is to modify the UnrealBuildTool to include the linker flags. For this flag, I went into $/Programs/UnrealBuildTool/Linux/LinuxToolChain.cs, went to the function "LinkFiles", and changed this line

`string ExternalLibraries = "";`

to this

`string ExternalLibraries = " -lstdc++ -lsupc++ ";`

This was around line 1150 for me, but will vary slightly depending on which version of the UnrealBuildTool you have. Note that the unreal build tool needs to be build manually by right-clicking on the Unreal Build Tool project and selecting "Build" - it's not build by default when the entire solution is built. This passed the necessary linker flags and allowed me to package the project for linux. I was able to package on a windows 7 box and run on Ubuntu 16.04.

sytelus, 2017-12-15:
@mitchellspryn I did the steps to cross compile from Windows. One snag I hit was to generate binaries for MavLinkCom.a and rpc.a. These are the dependency that is compiled outside of Unreal build and does not exist on Windows machine. I'm wondering how you had generated these two files...

mitchellspryn, 2017-12-15:
I built the project on my Linux box, and copied the assemblies to the proper location in the plugin. I'm not sure if that's a scalable / general solution, but it worked in my case.

sytelus, 2018-04-12:
Should be resolved in recent rpclib upgrade

closed 2018-04-12 by sytelus

AloshkaD, 2018-06-08:
@MitchellSternke I followed your suggestion to edit"string ExternalLibraries = " -lstdc++ -lsupc++ ";" but that didn't work for me.  I recompiled for Linux from within unreal engine. What  Unreal Build Tool are you referring to?

mitchellspryn, 2018-06-08:
I think you might have quoted the wrong Mitchell...

The unreal build tool is the custom build tool that you use to build the unreal engine and the associated game projects. If you're trying to cross compile for linux, you should already have built the unreal engine from source, and have the unreal build tool project at$/Programs/UnrealBuildTool. You can see if the linker flags are actually getting passed by adding Console.WriteLine() statements into the build tool; they'll show up in the output when packaging the project.

AloshkaD, 2018-06-11:
@mitchellspryn Oooops, thanks man :). I've compiled UE from source because I previously had it installed using the game launcher and I also included "C:\airsim\v10_clang-5.0.0-centos7\" in var "LINUX_MULTIARCH_ROOT." 
I followed your recommendation of editing ExternalLibraries in LinuxToolChain.cs , which is for me is located in "D:\UnrealEngine\Engine\Source\Programs\UnrealBuildTool\Platform\Linux"  The onlything that I couldn't do is build the UnrealBuildTool . When I right click on it in VS I don't see a build option.

mitchellspryn, 2018-06-11:
Your project should look like the attached image. In that case, just right click on the C# project and click build.
[image]

If you don't see all those projects, you may still be using your downloaded build of the binary instead of the version you built from source. In that case, try switching the unreal engine version to your source build (close VS, right click on the .uproject, select "Switch Unreal Engine version.")

TritonSailor, 2018-06-11:
I ran into this same issue.  

Adding the code @sytelus suggested above to AirSim.cpp resolved it.

AloshkaD, 2018-06-12:
@mitchellspryn It successfully builds 
[image]

I updated clang v10 to v11 and changed the environment to point to the new clang. This helped me resolved errors with the crosstoolchain being old when I build the airsm simulation for linux. But at some point the build fails and I get this error below which I'm still googling how it should be fixed . 

[image]

AloshkaD, 2018-06-12:
@TritonSailor Thanks for the update.