microsoft/AirSim #604: cannot run build.cmd

samquatched, 2017-11-09:
I've tried running "build.cmd" on Developer Command Prompt 2015 and VS2015 x64 Native Tools Command Prompt. However, the build still fails. 

[image]

I've not modified the codes and followed the tutorials provided. Am I missing a file that I need to download or how can I rectify this?

raymondchua, 2017-11-09:
Can you show the whole log?

sytelus, 2017-11-09:
Yeah, please post whole build output.

`build > log.txt`

Also make sure you have VS2015 with Update 3.

samquatched, 2017-11-10:
[image]
[image]
[image]

Yes, I have VS2015 with update 3. Is this the log output?

raymondchua, 2017-11-10:
@samquatched , why did you use screenshots? Can't you copy and paste the text from the log?

samquatched, 2017-11-10:
@raymondchua  I don't have the log.txt file

abc19850508, 2017-11-13:
you can try to use CMAKE 3.7

closed 2017-11-13 by sytelus