microsoft/AirSim #607: Change the resolution of depth map and surface normals

waldomatcher, 2017-11-10:
Please excuse me for asking this question. I knew similar questions were already answered: https://github.com/Microsoft/AirSim/issues/270 and https://github.com/Microsoft/AirSim/issues/228

I want to collect some high-resolution images with ground-truth depth map and surface normal for some computer vision tasks. I was able to change the RGB image resolution by changing the "settings.json" configuration file. But the resolution of the depth map and surface normal is fixed at 256x144.

https://github.com/Microsoft/AirSim/issues/270 and https://github.com/Microsoft/AirSim/issues/228 suggested to modify Content/HUDAssets/FpvRenderTarget or DepthRenderTarget or SegRenderTarget uasset files. I don't have any experiences in Unreal. I don't know how to modify such settings, and I cannot find the specified file names in the AirSim/Unreal/Plugins/AirSim/Content/HUDAssets directories.

Can someone elaborate more on how should I change the resolution of depth maps? Or point me to some tutorials of how to achieve this in Unreal? Thanks.

Mayankm96, 2017-11-10:
Hey! 

With the recent changes in the simulator, the method specified in those issues are no longer useful. To set the size of the depth and normal maps you need to modify the `settings.json` file in the `~/Documents/AirSim` folder. 

Check the discussion on the issue [#527](https://github.com/Microsoft/AirSim/issues/527).

sytelus, 2017-11-13:
Add this in your settings.json ([see doc](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#image-capture-settings)):

```
   "CaptureSettings": [
    {
     "ImageType": 1,
      "Width": 640,
      "Height": 480
    },
   "CaptureSettings": [
    {
      "ImageType": 2,
      "Width": 640,
      "Height": 480
    },
   "CaptureSettings": [
    {
      "ImageType": 6,
      "Width": 640,
      "Height": 480
    }
```

closed 2017-11-13 by sytelus

majid1268, 2019-04-13:
Could you please share your settings.json file? 
I am also trying to change the default values from 255x144 into 640x448 (Width and height) but I was not successful.
I changed the settings.json file based on the doc in [https://github.com/Microsoft/AirSim/blob/master/docs/settings.md](URL) and checked the issue #527 but unfortunately, they could not help me and the values are still unchanged.
My settings.json file is: 

`{
"SettingsVersion": 1.2,
"CameraDefaults": {
"CaptureSettings": [
{
"ImageType": 0,
"Width": 640,
"Height": 448,
"FOV_Degrees": 70,
"AutoExposureSpeed": 100,
"MotionBlurAmount": 0
},
{
"ImageType": 2,
"Width": 640,
"Height": 448,
"FOV_Degrees": 70,
"AutoExposureSpeed": 100,
"MotionBlurAmount": 0
}
]
},
"PhysXCar": {
      "VehicleType": "PhysXCar",
      "Cameras": {"CaptureSettings": [
      {
      "ImageType": 0,
      "Width": 640,
      "Height": 448,
      "FOV_Degrees": 70,
      "AutoExposureSpeed": 100,
      "MotionBlurAmount": 0
      },
      {
      "ImageType": 2,
      "Width": 640,
      "Height": 448,
      "FOV_Degrees": 70,
      "AutoExposureSpeed": 100,
      "MotionBlurAmount": 0
      }
      ]
      }
    },
"SimMode": ""
}`

I will appreciate if you can help me to understand where is my problem.  

Best in advance