microsoft/AirSim #609: msgpackrpc not communicating

Jaleyhd, 2017-11-10:
File "camera.py", line 34, in <module>
    client.confirmConnection()
  File "/home/jaley/AirSim/PythonClient/AirSimClient.py", line 160, in confirmConnection
    home = self.getHomeGeoPoint()
  File "/home/jaley/AirSim/PythonClient/AirSimClient.py", line 169, in getHomeGeoPoint
    return GeoPoint.from_msgpack(self.client.call('getHomeGeoPoint'))
  File "/usr/local/lib/python3.5/dist-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/usr/local/lib/python3.5/dist-packages/msgpackrpc/future.py", line 43, in get
    raise self._error
msgpackrpc.error.TransportError: Retry connection over the limit

sytelus, 2017-11-13:
Are you running AirSim successfully? Looks like either its not running or its not allowed to open TCP/IP ports.

Terrywjx, 2017-11-30:
I also encountered this problem...

nibeh, 2017-11-30:
Are you using drone or car mode?

Jaleyhd, 2017-11-30:
Car mode . I suspect that with Ubuntu, there will be issues. I  will get back to you with more details. For now, I am closing the issue

closed 2017-11-30 by Jaleyhd

srivatsankrishnan, 2018-02-13:
@crazymuse I am seeing the same issue. Did you root cause this issue?

husha1993, 2018-04-18:
@crazymuse I am seeing the same issue. Did you root cause this issue?

TritonSailor, 2018-05-16:
I also experienced this issue running PythonClient on ubuntu over ssh with UE4 running on another ubuntu machine.  Running PythonClient locally, or over ssh on a windows machine, the PythonClient worked normally.  

I solved this issue by running `pip install msgpack-rpc-python` on the ubuntu machine running the ssh client. The PythonClient is working over linux ssh for me now.

sytelus, 2018-06-23:
Also see this: https://github.com/Microsoft/AirSim/issues/1141#issuecomment-399627155