microsoft/AirSim #610: Placing External Camera's to observe multiple vehicles

spencerh-b, 2017-11-10:
Is it possible to place stationary external camera's above the map and have them all record what is going on? I need at least two camera's recording from above. I tried to implement this by my self, but I was unable to figure out all the bugs.

1) Manual View Mode isn't working as a default.

2) Unable to find external camera if the Camera Director and BP camera are placed in the map and attached.

spencerh-b, 2017-11-10:
in the mean time i have just place two stationary multi rotors to provide images.

msb336, 2019-10-02:
You can do this by creating a [camera blueprint](https://docs.unrealengine.com/en-US/Gameplay/HowTo/UsingCameras/Blueprints/index.html) which tracks the player pawn (use the `getPlayerPawn` blueprint node). You can bind the camera view to a keybinding by opening the level blueprint, adding the camera actor to the blueprint,  and following the blueprint outlined in step nine of the above link. Instead of using `Event BeginPlay`, use a keybind event of your choice.

PR #3320 (2021-07-26): Add support for fixed external cameras

Fixes: #3208 
Fixes: #1896, 
Fixes: #1709, 
Fixes: #610, 
Fixes: #599
 

## About

This PR adds the capability to create a fixed camera separate from the cameras on the vehicle, such as a CCTV camera. I'm not aware of any simple way to go about doing this currently, and therefore thought to try it out. I haven't checked the issues very thoroughly, so please let me know if there are other related issues and ways to go about doing this.

PR has been opened to get ideas and suggestions and whether this is actually required. This will likely become a big PR, [... cut]
closed 2019-10-02 by msb336