microsoft/AirSim #611: location coordinates in unreal and airsim

raymondchua, 2017-11-10:
Hi, 
Is there a way to use the location information in Unreal for AirSim? How is the mapping like?
For example, I have a cone in Unreal at position 10,10,10. What values should I use so that my drone goes to the cone (10,10,10)? I have tried directly using 10,10,10 but it did not work.

sytelus, 2017-11-13:
I have updated the doc here: https://github.com/Microsoft/AirSim/blob/v1.1.5/docs/apis.md#coordinate-system

closed 2017-11-13 by sytelus

raymondchua, 2017-11-15:
@sytelus , Thanks! Works well!

lucafei, 2020-08-24:
> @sytelus , Thanks! Works well!

Hello, @raymondchua i have also same problem with you, can you show me how to transfer the postion from ue enviroment to airsim api?

datonefaridze, 2022-09-13:
@lucafei Hi did you mange to solve that? can you please help me?