microsoft/AirSim #613: Problem : airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A irLib\AirLib.vcxproj]

tanmoy-sabir, 2017-11-13:
Hello.

I am facing these problem while compiling build.cmd in Visual Studio 2017 Tools Command Prompt.
I have no idea how to solve this problem.

DroneControllerBase.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
MultirotorRpcLibServer.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
MultirotorRpcLibClient.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
CarRpcLibServer.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
CarRpcLibClient.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
SafetyEval.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
ObstacleMap.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
RpcLibServerBase.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]
RpcLibClientBase.cpp
c:\git\airsim\airlib\include\common\vectormath.hpp(12): fatal error C1083: Cann
ot open include file: 'Eigen/Dense': No such file or directory [C:\Git\AirSim\A
irLib\AirLib.vcxproj]

lovettchris, 2017-11-13:
try using "build.cmd" which will download the Eigen folder for you and put it in the right place.

sytelus, 2017-11-13:
If you are using VS2017, please use this thread for further discussions: https://github.com/Microsoft/AirSim/issues/595

closed 2017-11-13 by sytelus

wangshenfeng, 2018-06-03:
however， It doesn't work

AloshkaD, 2018-06-26:
Copying Eigen manually into Airlib/include could help compiling airsim but causes issues later with compiling projects. To solve this issue I installed VS 2015 as a component in my VS 2017. See attached screenshot
[image]

Nagule8, 2019-09-04:
I'am also having this issue pls help me.
[image]

lovettchris, 2019-11-09:
Did you run "build.cmd" which downloads Eigen from http://bitbucket.org/eigen/eigen/get/3.3.2.zip ?

caesar84, 2020-02-26:
hi there 
I am having the same problem even though I have tried both 
 x64 Native Tools Command Prompt for VS 2017
and 
Developer Command Prompt for VS 2017
[image]

lovettchris, 2020-02-26:
run `build.cmd` to get Eigen downloaded properly.

caesar84, 2020-02-26:
I am actually running it because I am following this guide [](https://microsoft.github.io/AirSim/docs/build_windows/)

caesar84, 2020-02-26:
restarted the pc and it went well this time with build.cmd.

yzexeter, 2020-10-18:
> Copying Eigen manually into Airlib/include could help compiling airsim but causes issues later with compiling projects. To solve this issue I installed VS 2015 as a component in my VS 2017. See attached screenshot
> [image]

This solves my problem. I am using windows v1.3.1, and the "build.cmd" can not download Eigen from bitbucket. I checked the url provided by @lovettchris , not existing anymore.

To those who have the same trouble, go to Eigen official website and download it manually. Unzip it and copy the Eigen folder(which is located under the eigen-X.X.X, version num, folder) into Airlib/include directory. Done.

CSUN1997, 2021-01-29:
I got the same issue and I found out that the bitbucket repo for Eigen has been moved to gitlab. Try change the link for Eigen to "https://gitlab.com/libeigen/eigen/-/archive/3.3.2/eigen-3.3.2.zip" and delete the cache and recompile.

Shubhamn96, 2021-04-10:
So I am also running into this issue. I am installing on the C drive with Visual Studio 2019. It worked on my personal laptop but now that I am trying to install it on to my work machine, it isn't working. I get this error

[image]

clovett, 2021-04-10:
Did you run "build.cmd" ?  build.cmd installs Eigen.

eherozhao, 2021-04-20:
I am using VS2017 and cannot build the Block project successfully. When I run build.cmd, the result is

`Hello there! We just upgraded AirSim to Unreal Engine 4.24 and Visual Studio 2019.
Here are few easy steps for upgrade so everything is new and shiny:
https://github.com/Microsoft/AirSim/blob/master/docs/unreal_upgrade.md`

Looks like it doesn't download libs online. How should I work it with VS 2017?

GhazalTajik1998, 2021-11-30:
So I had the same problem and was dealing with it for a day.I tried everything which is said here but none of them worked.

Then I used "x64 Native Tools Command Prompt for VS 2019" instead of "Developer Command Prompt for VS 2019" to run build.cmd.
it worked for me.

azamkamranian, 2022-01-29:
just copy engine3.zip files into ```\AirSim\AirLib\deps\eigen3```

Coinstine, 2023-02-07:
clear the cache(eigen3),and re-run build.cmd