microsoft/AirSim #614: Segmentation Issue in the Neighbourhood Environment

Mayankm96, 2017-11-13:
Hey all, 

I have been trying to provide the segmentation labels for the `landscape` and `sky` objects in the Unreal Neighboorhood environment. However, I am unable to set them to distinct colors while using the recent python APIs. I am referring to the [`segmentation.py`](https://github.com/Microsoft/AirSim/blob/master/PythonClient/segmentation.py) to do the same. I suppose the reason for that is these are not static meshes in the environment which might be causing this issue.

It would great if someone could help me out with fixing this problem. Thank you in advance!

sytelus, 2017-11-14:
This works to set color for sky:

```
found = client.simSetSegmentationObjectID("SkySphere", 42);
print("Done: %r" % (found))
```

closed 2017-11-18 by sytelus

Mayankm96, 2017-11-21:
@sytelus 

Thank you for the assistance for setting the sky color. However, I am still unable give the `Landscape` in the Neighborhood Environment a color. It seems that the `Landscape` is a different kind of [mesh](https://docs.unrealengine.com/latest/INT/Engine/Landscape/Editing/SculptMode/index.html) due to which it appears transparent as shown below:
[image]

For the scene input here:

[image]

The segmented output that I get is:
[image]