microsoft/AirSim #615: python program control error

PantelisKyriakidis, 2017-11-14:
Hello,

I installed msgpackrpc and first i opened airsim. 
Then I tried to run ''hello drone'' and i got this:
Traceback (most recent call last):
  File "main.py", line 2, in <module>
    from AirSimClient import *
ModuleNotFoundError: No module named 'AirSimClient'

And i just can't find a solution anywhere so I guess I am missing something.
Anyone who can help ?

closed 2017-11-14 by PantelisKyriakidis

BrunoXL, 2018-04-02:
@PantelisKyriakidis  Have you solved it?

PantelisKyriakidis, 2018-04-04:
yeah i don't remember exactly but i was missing some .py file because i didn't download it all together

BrunoXL, 2018-04-04:
From where you downloaded the files? I `git clone` AirSim repository, so cannot be missing a file.

PantelisKyriakidis, 2018-04-05:
hmm then it's not the same error. I downloaded from the repository too.