microsoft/AirSim #616: Where is the binary executable file after build in Linux

zmonoid, 2017-11-15:
As titled

kunaltyagi, 2017-11-16:
I found them in _AirSim_/`build_debug/output/bin`

sytelus, 2017-11-18:
Yes, look in to build_debug folder. Below are the relevant lines from build.sh:

Below $build_dir is set to build_debug .

```
cp $build_dir/output/lib/libAirLib.a AirLib/lib
cp $build_dir/output/lib/libMavLinkCom.a AirLib/deps/MavLinkCom/lib
cp $build_dir/output/lib/libAirSim-rpclib.a AirLib/deps/rpclib/lib/librpc.a

# Update AirLib/lib, AirLib/deps, Plugins folders with new binaries
rsync -a --delete $build_dir/output/lib/ AirLib/lib/x64/Debug
rsync -a --delete external/rpclib/include AirLib/deps/rpclib
rsync -a --delete MavLinkCom/include AirLib/deps/MavLinkCom
rsync -a --delete AirLib Unreal/Plugins/AirSim/Source

# Update Blocks project
Unreal/Environments/Blocks/clean.sh
mkdir -p Unreal/Environments/Blocks/Plugins
rsync -a --delete Unreal/Plugins/AirSim Unreal/Environments/Blocks/Plugins

```

closed 2017-11-18 by sytelus