microsoft/AirSim #618: [Linux] error while loading shared libraries: libc++.so.1

kunaltyagi, 2017-11-16:
OS: Ubuntu 16.04

Make command: `./clean.sh && ./setup.sh && ./build.sh`

HEAD: d2dd11855

## Issue
Same issue with every binary file
```
$ ./build_debug/output/bin/MavLinkTest 
./build_debug/output/bin/MavLinkTest: error while loading shared libraries: libc++.so.1: cannot open shared object file: No such file or directory
```

Noticed the following in the build output:
```
clang: warning: argument unused during compilation: '-nostdinc++' [-Wunused-command-line-argument]
```
Persists on removal of `-stdlib=libc++` from cmake/cmake-modules/CommonSetup.cmake

yuyangch, 2017-11-17:
Same issue here, What is the correct compiler statement to use? If I want to compile a test.cpp for example
In the format of :
clang++-3.9  -I ? -L ?  test.cpp -o test

sytelus, 2017-11-18:
I'm wondering you might have something left behind from previous build. I did the following and everything completed without error:

```
git clean -fdx
git pull
./setup.sh
./build.sh
```
BTW, I noticed that `git clean -fdx` sometime fails to remove llvm-* folders. So you might have to remove them manually. If you know why that happens, it would be good to know :).

kunaltyagi, 2017-11-20:
@sytelus The compilation is not an issue. I did follow your steps and had the same errors while trying to run the binaries. Can we put something in the docs related to running them?

Thanks to @yuyangch , my fix is
```
$ LD_LIBRARY_PATH=./llvm-build/output/lib/:$LD_LIBRARY_PATH ./build_debug/output/bin/<binary> <args...>
```
I don't need sudo privileges for this. Does anyone have something a little better than this?

PR #622 (2017-11-22): Hard-coded path for run-time search of libc++.so.1
Line 78: Added `rpath` hint to assist in run-time linking

Rest all are whitespace pruning

Fixes #618 for novices. Enables simpler running from cmdline and GUI without polluting `LD_LIBRARY_PATH`
closed 2017-11-22 by sytelus