microsoft/AirSim #621: Build LIDAR capabilities in AirSim for Cars.

shrimalmadhur, 2017-11-19:
I'll be adding LIDAR capabilities for the Car model. 
ETA: 11/4

mitchellspryn, 2017-11-20:
@shrimalmadhur Did you mean 12/4? It's currently 11/19...

shrimalmadhur, 2017-11-20:
Thanks. Updated!

sytelus, 2017-11-22:
Thanks. It would be great if you can summarize/outline your design.

shrimalmadhur, 2017-11-22:
I saw your comments in some other Issue that says to extend the SensorBase class the same way other Sensors have extended it. And then I can use the Raycasting function which is already there in UAirBluePrintLib. I am planning to do it in the same way.

tejaskhot, 2017-11-27:
Would this not be available for use with the drone?

sytelus, 2017-11-28:
@tejaskhot yes. All AirSim sensors are reusable across vehicle types.

nnayeer, 2017-12-06:
Any updates? Eagerly awaiting this.

shrimalmadhur, 2017-12-06:
Got caught up in other stuff. Will update the issue once I have some update.

realdealneil, 2017-12-15:
I am particularly interested in adding a laser altimeter to the simulator (or at least having ground truth for height-above-terrain).  I'm guessing that's a subset of what you're planning to do?

tejaskhot, 2018-01-10:
Is there any update on this?

benasena, 2018-01-30:
I have most of the pieces of this working. I did the ray trace in VehiclePawnWrapper, then passed it through MultiRotorConnector to a variable added to environment to get it in a distance sensor class based on the barometer implementation. I added the necessary reporters and MAVLink handlers, and can see the measurement in QGroundControl in the analyze widget and if I run "listener distance_sensor" in nsh it all looks good. I set EKF2_HGT_MODE=2 and it flies well in all modes.

For some reason it still doesn't seem to follow terrain, though. Am I missing parameters or is there something about the HIL firmware that ignores distance sensors? I believe I correctly implemented the members of the DISTANCE_SENSOR MAVLink message (http://mavlink.org/messages/common). My units are all centimeters there and I get meters when I run the nsh command. I did use Unix epoch time instead of boot time. Could that be it? Is there a good way to an appropriate value for that in AirSim?

nnayeer, 2018-01-30:
I work mainly on ROS, but I've managed to get a pointmap stream using depthperspective images (passing it through the appropriate conversions and ros_perception/depth_image_proc). This feels good for ground truth data but maybe some filters and noise would make it more like an actual lidar? Would be especially handy if I could spawn 4 cameras at the same point (rotated 90 degrees) and get a 360 degree pointcloud.

benasena, 2018-02-01:
Hm, so when I set the output of the distance sensor to constant 0 and had EKF2_HGT_MODE set to use it, it seemed to fall back on the barometer so I'm not convinced it was ever using the distance sensor. I'll have to check the PX4 firmware code to see how this needs to be integrated I guess.

Interestingly, if I set it to computer vision height mode, it just rejected altitude control instead of falling back.

benasena, 2018-02-01:
Looks like it gets copied here:
https://github.com/PX4/Firmware/blob/master/src/modules/ekf2/ekf2_main.cpp#L386
	_vdist_sensor_type(this, "HGT_MODE", true, _params->vdist_sensor_type)

And is used here to determine the mode with fallbacks.
https://github.com/PX4/ecl/blob/master/EKF/control.cpp

It does look like it only rejects it based on timeout, so I'll see about using the boot time somehow rather than epoch time.

zergler, 2018-02-23:
Thanks, any updates on this? For what I'm doing, it's very important to get as close as possible to the real deal, so the simulated LIDAR should be configurable and be able to model different LIDARs. Some of the parameters I see this sensor having are:

- Min and max laser range
- Number of lasers per sensor array
- Rotation speed of LIDAR sensor array
- Number of laser sets (usually 2)
- Vertical and horizontal field of view for each set
- Angle between each set
- Spacing between each set

curly, 2018-02-23:
Hm, what I did was just for a fixed single laser. A scanning LIDAR shouldn't be much harder to implement, though there are more parameters to accommodate for specific implementations. In general, the rotation can be accomplished with modular arithmetic on UGameplayStatics::GetTimeSeconds(GetWorld()).

benasena, 2018-04-09:
I finally got around to testing the utilization of the rangefinder for height estimation in SITL and figured out why it wasn't using it and reacting to terrain. Turns out I was using inav instead of ekf2. Whoops. I tested in SITL with SYS_MC_EST_GROUP and other parameters set appropriately and it worked with very clear terrain reaction. I'll test soon with HIL and confirm it was the problem there as well, then see if I need to change the way timestamps are created to get better performance.

jeremypjoseph, 2018-06-22:
any update on adding scanning lidar to airsim?

DrifterFun, 2018-08-03:
I notice that a similar system called [Carla](https://github.com/carla-simulator/carla) has Lidar ability. Is it posiible to transplant the codes into AirSim？

closed 2018-10-08 by sytelus

zhoudacheng6, 2021-01-03:
Hey bro，i want to know to get lidar data.I saw the getLidaData() function on the document, but I don’t know how to run it

PR #1424 (2018-10-08): Add support for Lidar sensor
Add Lidar sensor in AirLib object model as well as Unreal
implementation using ray-casting. Add python client example.

Future work:
    - Enable configuration of lidar parameters
    - Enable visualization of point-cloud on client side

Resolves: #621
commit 88d2ee6d (2018-10-08): Add support for Lidar sensor

Add Lidar sensor in AirLib object model as well as Unreal
implementation using ray-casting. Add python client example.

Future work:
    - Enable configuration of lidar parameters
    - Enable visualization of point-cloud on client side

Resolves: #621
commit 9986414b (2018-10-08): Add support for Lidar sensor

Add Lidar sensor in AirLib object model as well as Unreal
implementation using ray-casting. Add python client examples.

Future work:
    - Enable configuration of lidar parameters
    - Enable visualization of point-cloud on client side

Resolves: #621