microsoft/AirSim #626: Unable to set segmentation color to `landscape` in Neighborhood Environment

Mayankm96, 2017-11-21:
This is an extension to the issue #614:

In the Neighborhood Environment, it seems that the `Landscape` is a different kind of [uasset](https://docs.unrealengine.com/latest/INT/Engine/Landscape/Editing/SculptMode/index.html) in which the grass is actually a [material](https://docs.unrealengine.com/latest/INT/Engine/Rendering/Materials/). Due to this, it appears transparent when viewed from below (as shown below):

[image]

When I try to set the `landscape` a segmentation color by running the statement
``` 
success = client.simSetSegmentationObjectID("landscape[\w]*", label["TERRAIN"], True);
```
the grass part in the object is not given the color specified and it appears as if it is not there in the segmented image. For instance, for the scene input here:
[image]

The segmented output is  (here, the blue is the color I have set to the `SkySphere`, while the landscape was set to a brown color):
[image]

Any idea on how this could be fixed?

sytelus, 2017-11-23:
I investigated this. There are two separate issues. First AirSim code didn't handled Landscape type objects which happens to need special treatment. I've now [added this code](https://github.com/Microsoft/AirSim/commit/1dd5956fcb1948c5aaa14e2aa1c1e134837e06b2). But still the issue is another bug in Unreal Engine that causes no update in rendering even after setting object ID on Landscape. So while new changes are there and your above code will indeed return success == true, rendering still doesn't reflect it. I have posted this issue on [Unreal forums](https://answers.unrealengine.com/questions/728584/landscape-type-objects-dont-reflect-changes-in-cus.html) as well.

Fortunately, I did eventually found the workaround. If you select Landscape in Unreal Editor and set "CustomDepth Stencil Value" in the editor then it does work. I'll do this for the environments that we have in releases. If you have your own environment then you can set this as shown in below image (also get latest code from git). Search for landscape object in World Outliner, go to Rendering section in details, set "CustomDepth Stencil Value" to your choice between 0 to 255. Check the box "Render CustomDepth Pass".

[image]

Mayankm96, 2017-11-25:
Oh, I see. This helps a lot! Thank you for the fix. :)

Would you like me to close this issue, or wait till the bug is fixed?

closed 2018-07-03 by Mayankm96