microsoft/AirSim #627: How do I use my own car model?

n1ckfg, 2017-11-21:
Looking to change the specs of the default car model--is there anything addressing this in the documentation?

mitchellspryn, 2017-11-22:
If you're just looking to change the physics and are OK with keeping the same visual model, the simplest thing would be to edit the parameters of the included uasset file for the SUV (in AirSim Content/VehicleAdv/SUV/)

If you have a completely custom mesh, then you can set it up like any other wheeled vehicle, but instead parent it to CarPawn. And then change the CarPawn constructor to point to the correct assets.

n1ckfg, 2017-11-22:
Thanks! Yup, just looking to change the physics, don't care about the appearance of the geo for now. Have you got anything there that would affect the throttle command? (Is there a drivetrain model?)

mitchellspryn, 2017-11-22:
For throttle, there are a few things you can try. 

First, open the blueprint SuvCarPawn, and navigate to the VehicleMovement component:

[image]

There are a few relevant parameters you can play with:

- Max RPM: this controls the top speed of the car. 
- Torque Curve: this controls the acceleration. To up it, add a few more points and slide it up. 

There are drivetrain / transmission models, but those usually affect handling (i.e. turning) more than acceleration. 

Do keep in mind that there will need to be some manual tuning; I have found that copy-pasting actual data from real-world tests done on car components (e.g. torque curves from engines) usually doesn't give very good results.

n1ckfg, 2017-11-29:
Awesome, I'll play with this!

closed 2017-11-29 by n1ckfg

benjamin-lange, 2018-03-16:
Would you please provide a more detailed tutorial on how add a custom mesh vehicle to AirSim? Thank you!

chioveri, 2018-04-09:
Hi, I would also be happy to have more infos on how to use a custom mesh car/vehicle. Thank you

benjamin-lange, 2018-04-19:
@chioveri 

May be this works for you, I will try it out as soon as I find the time:
https://github.com/Microsoft/AirSim/blob/master/docs/settings.md
The documentation deals with all sorts of parameters that can be
set in the "/Documents/Airsim/settings.json file". Have a look at the 
"pawn paths" section.

fernandodamasio, 2018-10-19:
@mitchellspryn 

About your comment:

> If you have a completely custom mesh, then you can set it up like any other wheeled vehicle, but instead parent it to CarPawn. And then change the CarPawn constructor to point to the correct assets.

Do we change the mesh in this code snippet here?

```
ACarPawn::ACarPawn()
{
    static ConstructorHelpers::FClassFinder<APIPCamera> pip_camera_class(TEXT("Blueprint'/AirSim/Blueprints/BP_PIPCamera'"));
    pip_camera_class_ = pip_camera_class.Succeeded() ? pip_camera_class.Class : nullptr;

    const auto& car_mesh_paths = AirSimSettings::singleton().pawn_paths["DefaultCar"];
```

Do you have any tips on how to do it? We need to change the mesh of the car, using another blueprint that I have.

Best Regards,
Fernando Damasio

benjamin-lange, 2018-10-22:
@fernandodamasio
I did not look into your code problem.
But here is a skeleton settings.json for the custom model I use:
`
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "PawnPaths": {
	"DefaultCar": {"PawnBP":"Class'/Game/TruckTrailer/Truck/Blueprints/Truck.Truck_C'"}
	} 
`
You have to adjust the path in "DefaultCar" to your setup.
Please refer to the [documentation](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md) for details. Look under "Other Settings" paragraph "PawnPaths".

fernandodamasio, 2018-10-23:
> @fernandodamasio
> I did not look into your code problem.
> But here is a skeleton settings.json for the custom model I use:
> `"SettingsVersion": 1.2, "SimMode": "Car", "PawnPaths": { "DefaultCar": [...]

Hi @benjamin-lange 

I tried to change only the mesh inside the cars' blueprint. It didn´t work. So, I guess I need another car to change it? I mean, I can't change just the mesh / appearance of the Airsim car.

Appreciate your help.

Best regards,
Fernando Damasio

benjamin-lange, 2018-10-24:
> I tried to change only the mesh inside the cars' blueprint. It didn´t work.
I recommend leaving the default blueprint alone.
Just to clarify what you tried to do, you tried to change the mesh as it is shown in the following screenshot?

[image]

If so then you might break the animations associated with the mesh which could cause more problems.
That is if you want to use the default vehicle in the future you have to fix these problems. 

> So, I guess I need another car to change it?
Yes, I recommend that. At least copy the original Blueprint located in project_name\Plugins\AirSim\Content\VehicleAdv\SUV and rename it before you experiment on it. ;-)
After you made your changes try out the pawn path method as stated above.

For your experiments with custom vehicle Blueprints in Airsim have a look into this [free asset ](https://www.unrealengine.com/marketplace/carsim-vehicle-dynamics) from the marktplace  or simply use the car model from the Blueprint vehicle template.

fernandodamasio, 2018-10-24:
Hi @benjamin-lange 

Thank you very much for your quick response and also congrats for the Airsim project!

Regards,

benjamin-lange, 2019-01-02:
@karanparikh91, have you tried using the "WASD" keys to move the vehicle?

karanparikh91, 2019-01-11:
@benjamin-lange  thanks for your reply.

Turns out the car was not exported properly from Blender. 
My colleague found this blog post useful in fixing the problem
blog post: https://forums.unrealengine.com/development-discussion/content-creation/66676-help-blender-vehicle-to-ue4

I've got things working correctly now.

thejeshk, 2020-02-23:
Hi @benjamin-lange 
[image]

I have my own own project and my own vehicle model with the following skeleton [see image]. I have created a child blueprint class from AIRSIM carpawn and selected a mesh of my own vehicle. What else I should do in addition to that to use my vehicle ?  

If I try to play. Iam getting the following error. [image] 
[image] 

[image]

My json settings are follows. 

{
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "PawnPaths": {
    "DefaultCar": { "PawnBP": "Class'/Game/Mower/Blueprints/Mower_AIRSIM.Mower_AIRSIM'" }
  }
}

thejeshk, 2020-02-23:
Hello @benjamin-lange, @karanparikh91 @mitchellspryn @fernandodamasio > 

I am having difficulties adding custom vehicle into my own project. 

Here's what I have done so far: Followed same steps as @karanparikh91 

**I have created my own project** 

1. Imported my new fbx asset for the car. **It has the same skeletal hierarchy but different bone names.** 
[Note: Is it important to have the same name bone names?] 
2. Has physics asset. Wheels are set to kinematic bodies. I'm using single hull convex collisions on all bodies for my mesh
3. Created a BP for my new skeletal mesh for the car. The BP I created has the CarPawn as its parent
4. Changed the Airsim settings.json file to include the pawn path to my new blueprint

When I hit play, I do see my new car flying from somewhere and dropping down (even though the player start is on the floor) But it doesn't seem to respond to any input when I hit the arrow keys. Am I missing a step or something else?

Thanks!
TK