microsoft/AirSim #628: Unreal Engine Performance

brean, 2017-11-22:
Hi,

I got AirSim running on my notebook (Lenovo T470p, gtx 940mx graphics card). I know that this is not the best machine for gaming so I am getting only a few fps in AirSim. I already created an [ini file](https://gist.github.com/brean/d12f5fa72905e149c3d18d768540df3b) and lowered the resolution so that I only have a small window and it helped a lot but I am still a few frames away from smooth 30fps. I am just an Unreal Engine beginner, so maybe someone here has some hints on how to configure the simulation so it runs on my notebook (disable post-processing stuff, other places to lower quality)?

n1ckfg, 2017-11-22:
Look up Post-Processing Volume for instructions on how to turn off post effects. Lots of effects are set to on by default, and you can squeeze a lot more performance out of a project by disabling them.

brean, 2017-11-22:
Thank you! It's still not very smooth but at least a bit faster. When I am not recording I can drive around now. I guess I just need more GPU power ;-)

mitchellspryn, 2017-11-22:
If you're only using it for a little bit, I can confirm that AirSim runs on [Azure NV VMs](https://buildazure.com/2016/08/09/azure-n-series-vms-and-nvidia-gpus-in-the-cloud/). They do get pricey if you are planning on running extended experiments, though.

sytelus, 2017-11-22:
BTW, which environment are you running? Here are some more tips to get better perf:

- Turn off subwindows (if you had turned them on)
- Use binaries instead of running from code (usually 2X FPS)
- Use run.bat to run in Windowed mode and adjust the size of window
- Use smaller environments

brean, 2017-11-23:
I am using the LandscapeMountains-environment. It had the lowest filesize so I assumed that it is the smallest. But if there is another landscape that is smaller/is made of less polygons I could switch to that.

brean, 2017-11-29:
I compiled AirSim myself and now I am able to set the quality settings in the unreal editor to medium/low which gives me some more perfomance (because I have the whole Editor I can still tweak a lot but I'll close this issue for now).

closed 2017-11-29 by brean

n1ckfg, 2017-11-29:
Some of the examples are pretty intense--I find the current City demo chugs on an i7 6700K / Titan X. I'd run Blocks first to get baseline performance on your system; that should run well on a newish laptop.