microsoft/AirSim #630: Can not cross bridge

brean, 2017-11-22:
In the current LandscapeMountains-scene I can not cross the bridge and instead crash into it when I try to cross it with the car.

mitchellspryn, 2017-11-22:
Oh, I ran into this issue before when using the environment in my project. The problem is that the collision hitbox for the bridge mesh is set to be a simple rectangle. The asset needs to be modified to use a convex hitbox. 

@sytelus I can share the asset that I use. You should be able to drop it into the project and build, if that's the issue.

sytelus, 2017-11-23:
I've taked @mitchellspryn patch and uploaded new binaries for Mountain environment. This is free environment so you should be able use this patch by yourself as well.

closed 2017-11-23 by sytelus

hmishra2250, 2018-02-02:
For anyone seeking solution to this, remove the existing collision mesh from *SM_Bridge_A* asset and rebuild collision using high accuracy (I used 1) and 26 K-DOP setting. More on this [Here](https://docs.unrealengine.com/latest/INT/Engine/Content/Types/StaticMeshes/HowTo/SettingCollision/)